vitotai/BrewManiacEsp8266 #4: Compilation errors

allthepies, 2017-05-18:
Hi Vito,

I'm trying to compile the latest BrewManiacESP8266 code from Git and have hit a compilation error I can't resolve.
`In file included from sketch\BrewManiac.cpp:17:0:

pins.h:75: error: 'PCF8574' does not name a type

 PCF8574 pcf8574(PCF8574_ADDRESS,I2C_SDA, I2C_SCL);

 ^`

pins.h references a class called PCF8574 but the library you listed as required ( https://github.com/WereCatf/PCF8574_ESP ) doesn't define a class with that name, the class that is defined in pcf8574_esp.h is named "PCF857x".

I can't see how this would work.  Can you check please ?

allthepies, 2017-05-18:
This is hard work Vito !   I got past the above issue by installing the PCF8574 library from your BrewPiLess libs.zip file.   Now I get  the following.
`n file included from sketch\BrewManiac.cpp:311:0:

sketch\ui.h: In function 'void uiScanLcdAddress()':

ui.h:288: error: 'class LiquidCrystal_I2C' has no member named 'updateAddress'

     lcd.updateAddress(address);

         ^

exit status 1
'class LiquidCrystal_I2C' has no member named 'updateAddress'`

The library list you include in the BrewManiac libs/README needs a revision I think, the current set just doesn't work.

vitotai, 2017-05-18:
Please use the libraries I provided, and update to newer version if you know it would work better.

Some of them are modified while some of them might have changed and might break my code.

I try me best not to modify the library. However, sometimes I have to.
The LCD library is one good example. I like the idea of auto scanning of LCD address, but the address of LCD library is a private, not public nor protected, variable.

BTW, I've tried using newer versions of ESPAsyncTCP & ESPAsyncWebServer. The newer versions somehow are not as stable as old version. I don't have time to dig into it, so I keep using old versions.

allthepies, 2017-05-18:
OK, I didn't spot the libs.zip file.  When I copy the PCF, LCD and WiFiManager lib from that then it compiles.

Thanks for the reply, might be worth adding a comment in the Library README ?

closed 2017-05-31 by allthepies