vitotai/BrewManiacEsp8266 #5: Can't save recipes

allthepies, 2017-05-24:
Love the s/w Vito, got it up and running on a NodeMCU.

However, when I load in a beerXML file it renders fine but attempting to save it gives an alert popup with the text "Error saving recipe:500".  The serial output at the same time is "File UL Failed:/R/CitraGold"

BeerXML file attached.

[citra.xml.txt](https://github.com/vitotai/BrewManiacEsp8266/files/1026643/citra.xml.txt)

vitotai, 2017-05-26:
The code just opens the file "/R/CitraGold" and writes the content. Therefore, the original file, if exists, will be overwritten. 

I've tested the XML. Just like what I have guessed, there might be something wrong with SPIFFS. That means that you might need to format the SPIFFS and try again.

allthepies, 2017-05-26:
Reformatted SPIFFS and after restart + reloading of all the web content I get exactly the same result, the 500 error and same error in the serial log.

I don't understand how the handleUpload function is being invoked, this seems to create the file in SPIFFS but how is it invoked ?   I can see how the 500 error I'm seeing occurs, from the following code, invoked when a the UPLOAD_PATH URL is requested.

`		}else if(request->url() == UPLOAD_PATH){
			if( request->hasParam("file",true,true)){
				String file=request->getParam("file", true, true)->value();
				if(accessAllow(file,WRITE_MASK)){
					if( SPIFFS.exists(file)){
						DBG_PRINTF("File UL success:%s\n",file.c_str());
						request->send(200, "", "{}");
					}else{
						DBG_PRINTF("File UL Failed:%s\n",file.c_str());
						request->send(500);
					}
				}else{
					request->send(401);
					DBG_PRINTF("File UL not allowed:%s\n",file.c_str());
				}
			}else{
				DBG_PRINTF("miss file in req.");
				request->send(404);
			}`

That code doesn't actually seem to upload anything, it just checks to see if the file already exists and if it doesn't then returns a 500 error.

Is this the right code version ?

allthepies, 2017-05-26:
Yes, that's it.  The ESPAsyncWebServer version I have doesn't invoke the uploadRequest method when I click on the save button.

vitotai, 2017-05-26:
That's odd.
I found that adapting new library introduced new issues, so I ended up keeping old version.

allthepies, 2017-05-29:
Vito, I've reverted to using your library copies of the Async TCP lib + Webserver and now recipe saving works!

I do have a recipe based question though.

When I load a beerXML recipe in which has the strike (mash-in) temperature identified in the <INFUSE_TEMP> mash tag then BM is ignoring this value and calculating the mash-in temperature as per the recipe options settings.   Can't you include an option to use the beerXML supplied value ?

vitotai, 2017-05-30:
INFUSE_TEMP is optional. It's text type and more like for display. It can be done, easily, though. Ok. I've already done it by using a preference option.

allthepies, 2017-05-30:
Great, thanks!  That will be useful for people like me who create the whole recipe / mash profile in s/w like BeerSmith and then export the BeerXML into BM.  Everything is then set as it was in BeerSmith.

allthepies, 2017-05-31:
Fixed in v0.2.5 of web components.

closed 2017-05-31 by allthepies