vitotai/BrewManiacEsp8266 #7: Delete recipe doesn't work

allthepies, 2017-05-29:
I can now save recipes, which is great!

However, the delete recipe option doesn't work for me.  I click on delete and nothing happens.

Looking at the bm.js Javascript it would appear that the function associated with the "delRecipe" tag is incorrect and is using the output from "var r = t.viewing;" as the recipe name to be deleted whereas this actually returns an object.

The Chrome JS console gives the following error when I select "Delete"
`error condition
lib.js:2 Uncaught Error: Syntax error, unrecognized expression: #recipe-list-[object Object]
    at Function.fa.error (lib.js:2)
    at fa.tokenize (lib.js:2)
    at fa.select (lib.js:2)
    at Function.fa [as find] (lib.js:2)
    at n.fn.init.find (lib.js:2)
    at n.fn.init (lib.js:2)
    at n (lib.js:2)
    at rmList (bm.js:3332)
    at Object.success (bm.js:3341)
    at i (lib.js:2)`

And the Serial output from the ESP returns

```
RM request:R/[object Object]
RM executed:/R/[object Object]
```

So it looks like an Object is being used instead of a filename.

vitotai, 2017-05-30:
I couldn't believe that I didn't find this bug. The deletion of recipe seems unfinished.
It is fixed, and I will update later.

allthepies, 2017-05-30:
Thanks Vito. I tried to fix it just now but couldn't find an elegant way of determining the filename of the selected recipe (to be deleted).

allthepies, 2017-05-31:
Now fixed in v0.2.5 of web components.

closed 2017-05-31 by allthepies