vitotai/BrewManiacEsp8266 #10: Add Edimax SP-1101W Smart Plug

kabelbrand, 2017-07-18:
Please add the option to use the Edimax WLAN plug to switch on/off the heater for example. If you need some hints, please look at the CBP-project. Thanks!

kabelbrand, 2017-07-30:
Sad you closed my issue. I think using a WLAN plug is an interesting alternative for using a SSR, isn't it!?
I only use my WLAN plugs instead of SSRs, it's safer and easier and i think the implementation is not hard, if you have a look, how this works...

vitotai, 2017-07-31:
I don't think its a good idea. Not interested.

allthepies, 2017-07-31:
Do the WLAN plugs use a mechanical relay ?  If so then not a good choice for an application which uses a PID approach with short on/off pulses.

kabelbrand, 2017-07-31:
Hmmm... i'm not sure, but i use them since a lot of brews and they work perfect. If you use wrong parameters, the mechanical relay can be damaged, but then your parameters are set wrong;)

Would be a nice feature for those, who won't take a hand on high voltage parts (as me)...

vitotai, 2017-07-31:
No argument wanted. Just totally not interested. 

Something I found:
http://www.intercity-vpn.de/files/edimax-1101w/

BTW, I googled CBP to get nothing. That's the reason I closed it. Sorry about it. 
Again, I am not interested. Please do it by your self. Thanks.

kabelbrand, 2017-08-01:
Sorry for that. You can have a look at CPB  here -> http://web.craftbeerpi.com/

I only mentioned this because i thought it could be interesting. Would i have known before,
that you don't like to see beyond your own nose, i wouldn't have opened this issue!

closed 2017-08-01 by kabelbrand

vitotai, 2017-08-01:
I have limited time and resource, so I have to choose what to do.

I really don't think it's a good idea because of the mechanical relay and the nature of wireless network. Most of the wireless switches use mechanical relays, or they would be bulkier and  need to use heatsink. I don't think Edimax 1101w is an exception, but I still tried my best to google and find out the photo to prove my guess. Did you see the photo link I provided that answered a question for you?

CraftBeerPi is a good project, but it won't show up at the first page of the result of googling CBP.  In fact, I always suggest people to use craftbeerpi when they need more power and flexibility. There are a lot of solutions with different advantages. You should just pick one that meets your requirements, or you can CREATE one if you can't find any. 

You might think that adding a new feature is easy, but it's not true for me. Although this project is not created for business purpose and it is sort of casual, I still tried my best following the the software development process, which includes
-creating use cases
-design
-coding
-uni-test
-integration test

Since I have limited time and resource, I can do what seems feasible and useful to most people. In fact, the development of this project is slowing down because there are already features that I can't test, including dual heater, sparge heating, distilling, and stirrer.

I certainly will make or am making wrong decisions, but, well. who cares? The source is open, and there are still a lot of other solutions. Brewers will find they own ways.

allthepies, 2017-08-01:
I'm with Vito here.  To add capability to Wifi sockets may well push the ESP8266 too far, I suspect it would run out of RAM.  Plus a mechanical relay isn't a sensible choice for BMEsp given its use of PID + PWM heating control.  As Vito says, the source code is here for anyone to fork a copy and adopt to meet their own requirements.  Give it a go.