vitotai/BrewManiacEsp8266 #12: 'PCF8574' does not name a type

arudmin, 2017-08-11:
[screenshot](https://www.dropbox.com/s/b5t6rypru6h4pg9/Screenshot%202017-08-11%2018.33.55.jpg?dl=0)

Arduino: 1.8.1 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from sketch/BrewManiac.cpp:17:0:
pins.h:75: error: 'PCF8574' does not name a type
 PCF8574 pcf8574(PCF8574_ADDRESS, I2C_SDA, I2C_SCL);
 ^
sketch/pins.h: In function 'void btnPrepareRead()':
pins.h:91: error: 'pcf8574' was not declared in this scope
  _portvalue=pcf8574.read8();
             ^
In file included from sketch/BrewManiac.cpp:313:0:
sketch/ui.h: In function 'void uiScanLcdAddress()':
ui.h:302: error: 'class LiquidCrystal_I2C' has no member named 'updateAddress'
     lcd.updateAddress(address);
         ^
exit status 1
'PCF8574' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

vitotai, 2017-08-13:
Download the libraries from here:
https://github.com/vitotai/BrewManiacEsp8266/tree/master/libs

Or, use PlatformIO and download this branch:
https://github.com/vitotai/BrewManiacEsp8266/tree/master/libs

I no longer use Arduino IDE.

allthepies, 2017-08-14:
https://github.com/vitotai/BrewManiacEsp8266/issues/4

Switch to PlatformIO :+1:    or use Vito's libraries.

arudmin, 2017-08-14:
Thanks, I've replaced LC and PCF8574 libs from archive and compiling done.

closed 2017-08-14 by arudmin

NixxBr, 2018-06-21:
Sorry for the old age topic, but i am receive the same error ("PCF8574 does not name a type"). The links for download the libraries was broken (page not found).

Please Vitotai, can you check ?

vitotai, 2018-06-21:
The libraries should be downloaded why you download the whole project, or it doesn't?
https://github.com/vitotai/BrewManiacEsp8266/tree/master/lib