vitotai/BrewManiacEsp8266 #13: Use Digispark Attiny85 instead pcf8574

jhnf, 2017-08-28:
Hello! Why do not you use Digispark Attiny85 instead of pcf8574?

vitotai, 2017-08-28:
PCF8574 is a I2C IO expander, Attiny85 seems to be a micro controller.
I don't understand what is your question.

jhnf, 2017-08-28:
To Attiny85, you can connect analog buttons and flash the microcontroller to transfer the state of the buttons on the i2c bus... Easier than connecting the library pcf8574. I can modify your code to support buttons > Attiny85 > i2c
For example: http://www.mauroalfieri.it/elettronica/attiny85-iic-wemos.html

vitotai, 2017-08-28:
We might have different view of easiness. PCF8574 is cheap and everywhere, I don't fell any difficulty. 

However, your are encouraged to do whatever you want to the code. That is the spirit and purpose of open source.