vitotai/BrewManiacEsp8266 #14: Not support Russian language

jhnf, 2017-08-30:
Library esp8266-oled-ssd1306 and all its fonts do not support Cyrillic. 
Translation is not possible.

arudmin, 2017-08-30:
Вот нашел описание доработки библиотеки: http://arduino.ru/forum/programmirovanie/rusifikatsiya-biblioteki-adafruit-gfx-i-vyvod-russkikh-bukv-na-displei-v-kodi

Напиши потом, если все работает.

jhnf, 2017-08-30:
Видел этот вариант. Пробовал. Это другая библиотека и формат шрифтов совершенно другой. Вообще все утилиты сборки шрифта работают по другому, не в том формате, в котором сделана текущая библиотека (эта вообще через зад формирует шрифт).
Остановился на библиотеке u8g2, нашёл русские шрифты и адекватную работу с дисплеями 1306 и 1106. В ближайшее время займусь адаптацией кода под эту библиотеку.

jhnf, 2017-08-31:
Выяснил преинтересную вещь! Используемая библиотка OLEDDisplay.cpp в принципе не понимает ничего, кроме английского. Там внутри зашита процедура, которая "отсекает" все символы с кодом более 127, то есть не английские. Переписал эту библиотеку - показывает все символы из шрифта, но в самих шрифтах нет кириллицы, а китайский генератор шрифтов не работает. Пробую переписать сам шрифт под русские символы.

Библиотека u8g2 так толком и не заработала, хотя на 3д принтере она без проблем русифицировала. Ковырять нет желания :)

arudmin, 2017-08-31:
Я использую обычный 2004 lcd, тоже хочу русифицировать библиотеку liquid crystal, используемую в проекте, так как она нестандартная. В один из вечеров попытался собрать, но не завелось. А почему ты решил олед этот использовать? Он же маленький и неудобно будет смотерть в него в процессе варки.

jhnf, 2017-09-01:
liquid crystal видел русифицированный, но там проблема в другом - китайские  2004 не содержат русских символов, поэтому они не русифицируются. У меня есть такой, сейчас Ardbir  стоит, но хочу по вафле варить.
ОЛЕД дисплеи очень яркие и даже мелкие буквы очень даже читабельные, пока запущу на нём, есть идея переделать интерфейс.

arudmin, 2017-09-02:
В Баварии установлен 2004 и он отображает русские символы. Проблема в используемой здесь библиотеке.

jhnf, 2017-09-04:
Проблема в дисплее и зашитой в него таблицы символов. Если у Вас 2004 от российского разработчика (который стоит в три раза дороже китайкого), то у него есть кириллица, в остальных вместо кириллицы иероглифы.
Думаете я на ardbir не пробовал русифицировать? Там уже изначально нормальная библиотека с поддержкой кириллицы, но с нормальным дисплеем...

vitotai, 2017-09-04:
I have been Russia in 2008 for tourism and tried learned the letters, but I forget all of them.

For the 20x4 LCD, it is possible to support Russian if you can find the one with Cyrillic font. 

(this one is not I2C version, but the I2C "board" is just a PCF8574 module, you can find it on eBay.)
https://www.aliexpress.com/item/Free-shipping-2pcs-Russian-cyrillic-font-Language-20x4-204-2004A-lcd-display-module-blue-professional-lcd/32809582876.html

Checking the specification of HD44780 specification, there are two version of ROMs: Japanese and European. The European font seems to have some Russian letter, but I can't tell.
http://fab.cba.mit.edu/classes/863.06/11.13/44780.pdf

For the OLED. we just need a library of font bitmaps and the decoding function to retrieve the bitmaps.

anonizmous, 2020-04-19:
> Выяснил преинтересную вещь! Используемая библиотка OLEDDisplay.cpp в принципе не понимает ничего, кроме английского. Там внутри зашита процедура, которая "отсекает" все символы с кодом более 127, то есть не английские.
> Пробую переписать сам шрифт под русские символы.

так удалось ли вам импортировать шрифты с русскими символами ?

jhnf, 2022-04-04:
> так удалось ли вам импортировать шрифты с русскими символами ?

Нет, не удалось. Забил на это дело. Переписываю код под сенсорный дисплей типа HMI и Nextion