vitotai/BrewManiacEsp8266 #15: Mash out setting what recipe imported

allthepies, 2017-10-22:
Hi Vito,

When importing a BeerXML recipe which has a mash out step defined, BMEx is adding another mash out step as per the recipe options settings.  Could you add an option in the recipes section to use mash out steps from the BeerXML ?  I know you did something similar for the mash-in temperature setting.

Thanks!   Myself and a brewing mate have built a Braumeister replacement controller using BMEx this weekend.  Works really well and my mate is very impressed.

vitotai, 2017-10-22:
What is the temperature of Mashout in your BeerXML?
If the temperature of last mash step is greater or equal to 75C, it is regarded as MashOut. 

Check this code:
`
if(mashes[mashes.length -1].temp >= 75){
		mashOut=true;
		mashes[mashes.length -1].aname = "Mash Out";
	}else
		mashOut=false;
	var step=1;
// [code deleted for better illustration]
	if(!mashOut)
		nmashes.push({aname:"Mash Out",temp:this.opt.MashOutTemp,time:this.opt.MashOutTime});
`

BrewManiacEx requires mandatory Mash Out rest. I followed the design of OpenArdBir, although the code was written from scratch.

allthepies, 2017-10-22:
Good question!  I can't remember.

So if BeerXML contains a step with a temp >- 75 degC then it will be treated as a Mash Out.  Will the "default" mash out as defined in the Recipes/options config of BMEx be ignored if a Mash Out is detected in BeerXML ?   I was getting the BeerXML mashout and then the additional mash out with BMEx had added.   Obviously I only want one.

vitotai, 2017-10-22:
yes. The default Mashout is inserted only when there's none in the BeerXML.

If you can provide me the BeerXML, it might help.

allthepies, 2017-10-22:
OK, I'll get hold of the BeerXML we used.

allthepies, 2017-10-22:
Mash out temp was 55 degC (we were testing)  so the s/w added the default mash out.

<INFUSE_TEMP>55.0 C</INFUSE_TEMP>

closed 2017-10-31 by allthepies