vitotai/BrewManiacEsp8266 #23: ESPAsyncTCP version in libs/ won't compile with recent platforimIO ESP8266 SDK

allthepies, 2018-01-03:
Having upgraded the PlatformIO ESP8266 SDK version recently, I then got compilation errors when building BMEsp.   

I believe the issue is this one:   https://github.com/me-no-dev/ESPAsyncTCP/issues/57

I updated the ESPAsyncTCP library to the latest version and BMEsp now compiles [I'm yet to test it].

vitotai, 2018-01-03:
I've tried to use latest version of ESPAsyncTCp and ESPAsyncWebServer.
There are some issues. The SSE Link will be closed and reconnected every 5 seconds. The result is that you can't edit the settings and automation process.
I am not sure wether it is caused by LWIP or ESPAsyncTCP. So far, it seems that LWIP doesn't report ACK from the peer. (I assume the TCP/IP of OSX behaviors as expected.)

allthepies, 2018-01-04:
Reverted to the previous platformIO esp8266 platform version and it now compiles OK with your versions of the ESPAsyncTCP libraries.

In platformio.ini modified
platform = espressif8266
to
platform = espressif8266@~1.5

allthepies, 2018-01-04:
Good stuff, I will try the latest version.

vitotai, 2018-01-05:
Yes.
I am trying WebSocket on BPL now.

vitotai, 2018-01-07:
The new libraries as well as WebSocket pass my stability test.

I also fixed some bugs:
https://github.com/vitotai/BrewManiacEsp8266/tree/v040pre

allthepies, 2018-01-07:
Just loaded up the v404pre branch and it's looking good.