vitotai/BrewManiacEsp8266 #27: Set point or pwn constantly changing after a UP+DWN reset

andreasferm, 2018-02-09:
I've noticed this while trying out the distill feature (again thanks Vito, it seems to be working. Still need to test the recipe and automation)

But if I do a up+dwn reset and go into manual mode for example the set point either keeps increasing or decreasing without pushing anything. Guessing that it's a logical hickup somewhere

vitotai, 2018-02-14:
I have to admit that I didn't test distilling well because I don't distill.

Although I can be very sure, your description seems to contradict the algorithm of keypad handling. The keypad event is processed when release except continuous pressing. The UP+DOWN logic is the same as all other mode. I can't find any issue by checking the code. Unfortunately, I won't be able to test on a real controller before end of March.

andreasferm, 2018-02-14:
It might be a browser issue.. I haven't tried with the keypad as I assembled everything while waiting for the gpio expander and am too lazy to connect the physical buttons. I'm not sure it correlates to the distilling feature. I will investigate a bit more late on.

vitotai, 2018-02-15:
Well, that makes sense. The function of continuous press is new. Will look into it later.

lekrom, 2019-11-10:
Hi Vito. I picked up the same issue. On my previous brew this occured and i first thought it was a stuck button. I disconnected the buttons and used the web interface. I switched off the power before disconnecting the button and resumed the brew afterward. The same thing happened on my next brew yesterday. The buttons were still disconnected. I then realised it must be a web interface issue. I also used the up dwn reset. I have distilling disabled. When i power cycled and resumed the brew all worked except the sparge heating stopped.