vitotai/BrewManiacEsp8266 #28: Temperature reached event occurred prematurely

allthepies, 2018-03-05:
Used BM v040pre for a real brew session last Saturday, all went well apart from one issue.

When in the boil stage, the temperature reached event fired when the temperature reached approx 86 degC.  This started the countdown timer etc and registered the event on the logging graph (attached below).  I pressed the pause button to suspend the timer and resumed once the boil had actually been reached.  So all ended well but had I not spotted the premature event trigger then the brew would not have worked out as planned.
The web UI did lose contact with BMEx a few minutes before, but reconnected OK.

[image]

vitotai, 2018-03-12:
Do you notice that I don't make latest framework default for BrewPiLess?

After migration to latest version, I setup a test platform of BPL which ran the latest framework and libraries. I eventually found that it was not as stable as old ones. BPL would become unreachable but ping-able after a period of time. It happened several times, but I couldn't find a way to reproduce it. It just happened after one day or longer. I am in a delimma of upgrading to latest framework or not. The continuous button operations of web interface will not be possible without WebSocket, which is not stable in old version.

allthepies, 2018-03-14:
Yes, I was aware of issues with the latest libraries.

However in this case I'm not sure that is the problem.  BM incorrectly entered boil temp reached state (when the actual temperature was just over 80 degC).  I would be surprised if that were down to the TCP/IP network libraries.

vitotai, 2018-03-15:
You are right. That should has nothing to do with TCP/IP or networking.
However, I don't remember I had touched anything around that part of code.
According to my debugging fu, I would guess it might be a memory corruption issue, which is the hardest type of bugs to solve.

vitotai, 2018-03-30:
If you check my code, you might find that I prefer using static global variables or local variables which are allotted in the stack. I try not using dynamic allocation to avoid memory fragmentation and commonly-seen pointer bugs.
It's not easy to reproduce this issue if it is truly memory corruption issue. If, however, it has something to do with the sensors, a simple low pass filer might do the trick. The question is the value of the filer parameter.

vitotai, 2018-06-11:
OK. So I had this issue, too. However, on BrewPiLess.
The temperature reading jumped to 100+ Celsius and then slowly went down because of the filters. It went up again, and I had to twist the connection of temperature sensors until it started to go down. After that, everything works fine. I have no idea why. The invalid value is supposed to be 85 degree Celsius. 100+ just doesn't make sense.
[image]