vitotai/BrewManiacEsp8266 #29: PID settings not updating

lekrom, 2018-03-11:
Hi Vito,

I am in the process of upgrading my brew kettle controller from the old Brewmaniac + BMESP8266 to the new BrewManiacEX. I have run into a few bugs.
The PID menu settings do not update correctly when using the LCD / button (offline) interface. I changed the updateSetting() function to add some debugging data to the code as follows:

bool updateSetting(int addr,byte value)
{
#if SerialDebug == true
  	Serial.print("Updating setting: ");
	Serial.print("addres: ");
	Serial.print(addr);
	Serial.print(" value: ");
	Serial.print(value);
	Serial.print("\n\r");
#endif		
	return SpiEEPROM.write(addr,value);
}

In the serial monitor, while attempting to change any setting, I see the following:
Updating setting: addres: 0 value: 64 
Updating setting: addres: 0 value: 11 
Updating setting: addres: 0 value: 33
Updating setting: addres: 0 value: 7
Updating setting: addres: 0 value: 17
Updating setting: addres: 0 value: 85
Updating setting: addres: 0 value: 35
Updating setting: addres: 0 value: 10
Updating setting: addres: 0 value: 50

The adress is always 0 regardless of the item I try to change. If I re-enter the PID setup menu, all values remain unchanged. 

Also, If I use the Web interface to change the settings, all works correctly except for the kD parameter which also does not respond here.

Any thoughts?

-Regards,
Lekrom

lekrom, 2018-03-11:
I just did another test, 

The "Unit parameter" and "Misc settings" menus does exactly the same as described above. The "set automation" menu appears to work correctly though.

-Lekrom

vitotai, 2018-03-11:
I might have find an issue, but I don’t have access to my controller before next week.
If you can build the code, add the following line at line2181@BrewManiac.cpp
‘_currentSettingAddress=_item.address;’

lekrom, 2018-03-11:
Thanx. I will give it a try and let you know.

vitotai, 2018-03-11:
The line is wrong.
find the definition of getItemDetail(void)
add the line at the end of the function

lekrom, 2018-03-11:
Fixed it, Thanks. You are a Rock star!

closed 2018-03-20 by lekrom