vitotai/BrewManiacEsp8266 #32: Sensor resolution setting

lekrom, 2018-03-17:
Hi Vito,

I noticed that the sensor resolution setting is not kept in Persistent Storage and thus the setting is lost on power down. Is there a reason for this? Probably not a huge issue but changes in sensor resolution may affect the PID auto-tuning accuracy. May be other complications as well.

-Lekrom

vitotai, 2018-03-19:
This is a design issue, which is a known issue that the designer doesn't want to handle it. :)

The resolution is be stored in the EEPROM of sensors. I am not sure why it is lost on power down. Maybe the sensor is defected or it has no EEPROM.

lekrom, 2018-03-19:
Hi Vito,

I had a look at the code and the DS18B20 datasheet, I think I found the bug. According to the datasheet, you are supposed to copy the scratchpad volatile version of the config register to the non-volatile EEprom config register (command 48h). I do not see you doing that in the code (correct me if I missed it).

Regards, Lekrom

vitotai, 2018-03-20:
Hi Lekrom,

Thank you for your information. I just copied the code of resolution setting code from somewhere on the internet without checking the document. I will check this later. 
Regards,
vito

lekrom, 2018-03-20:
Hi Vito,

I found the sollution, tested and it works:
in the void tpSetSensorResolution(byte *addr, byte res) function, add the foolwing at the end of the end of the function (after the ds.reset();)

    if(addr) ds.select(addr);
    else ds.skip();        
    ds.write(0x48);			// Save settings to non-volatile	
    delay(15);                            // Wait for EEPROM write to complete.

Regards, Lekrom