vitotai/BrewManiacEsp8266 #33: LCD/OLED display:  Pump indicator not restored after Iodine test

allthepies, 2018-05-13:
If the Iodine test phase is enabled then when exiting the iodine test phase to either mashout or extend mash then the pump indicator on the LCD isn't restored even though the pump is running.

The issue is that the pump on() method returns without doing anything if the pump is already on.  And so the pump indicator never gets restored.

I added a PumpControl method "updateUI"  and call this from the autoModeNextMashingStep() and autoModeIodineTestToMashExtension() functions

```
		void updateUI() {
			if ( isOn() ) {
				uiPumpStatus(PumpStatus_On);
		    wiReportPump(PumpStatus_On);
			} else {
				uiPumpStatus(PumpStatus_Off);
				wiReportPump(PumpStatus_Off);
			}
		}
```

autoModeNextMashingStep
```
	else if(_mashingStep ==7)
	{
		if(readSetting(PS_PumpOnMashOut)) {
			 pump.on();
		}
		else {
			pump.off();
		}
		pump.updateUI();

```

autoModeIodineTestToMashExtension
```
	setSettingTemperature( automation.stageTemperature(_mashingStep));

	#if	MANUAL_PUMP_MASH == true
	uiButtonLabel(ButtonLabel(Up_Down_PmPus_STP));
	#else
	uiButtonLabel(ButtonLabel(Up_Down_Pause_STP));
	#endif
    autoModeEnterMashingExtension();
    _mashingStageExtendEnable = false;
		pump.updateUI();
}
```

vitotai, 2018-05-15:
Thanks. Will apply your modification.

allthepies, 2018-05-15:
Thanks Vito.

One thing to note though, there may be other situations where this also occurs.

The issue happens when the pump is on then an event occurs which results in the screen being cleared.
When the screen is recreated the pump.on() call returns immediately (as the pump is already on) and so the pump symbol doesn't get drawn.

The exit from Iodine test event is one such case but there may be more.

BTW, do you have any more BMEx shields left ?  I'm looking for 1, possible 2.

closed 2018-09-03 by allthepies