vitotai/BrewManiacEsp8266 #37: Issue with sparge heating control.

lekrom, 2018-11-17:
Hi Vito. Found a bug with sparge heating. As soon as i enter auto mode, sparge heater is turned on and stays on. The lcd as well as lcd on web interface does not update the sparge temp in auto mode, frozen at last temp reading prior to entering auto mode. The temp #2 box on web interface does however seem to show correctly.

lekrom, 2018-11-17:
Sparge kettle was boiling @95 deg C while mashing. (typicall for my altitude) it was set to 75 (mash out)

lekrom, 2019-05-14:
Figured it out, I think. Think the eprom was not set up correctly after upgrading. Saving settings from lcd menu fixed it.

closed 2019-05-14 by lekrom

at001984, 2019-05-17:
how did you solve the problem?

lekrom, 2021-04-26:
This problem has popped up again. Not sure how to fix