vitotai/BrewManiacEsp8266 #39: Inverted Logic for Heater (using relay module in gas system)

gilson18, 2019-02-11:
Hi Vito, is it possible to have “inverted logic” option for heater, like for pump? 
I am using the same relay module, with inverted logic, but just pump works well

vitotai, 2019-02-13:
Relay modules are not recommended for heating control. Heating control switches faster and more frequent than pump control. Mechanical relays might react slowly and fail quickly.

If you still insist, change the line@
https://github.com/vitotai/BrewManiacEsp8266/blob/37abaa4f665191833882669f77456036f338f21d/src/pins.h#L118

from 
`digitalWrite (HeatControlPin, v);
`

to
`digitalWrite (HeatControlPin, (v==LOW)? HIGH:LOW);
`

gilson18, 2019-02-13:
Thank you very much.
I am using relay because I have a gas system, using valve instead of
heater, ssr doesn’t work well with valve. I have changed pid windows time.
If you have any other tip for this kind of system, would be great

Thank you again

vitotai, 2019-02-15:
Someone has done that, check this out:
https://www.homebrewtalk.com/forum/threads/yabc-yet-a-brew-controller-brewmaniacex.596907/page-5