vitotai/BrewManiacEsp8266 #40: LevelSensor "normal-open"

EroshKA, 2019-02-17:
I want to use a [normally open](https://ru.aliexpress.com/item/AC220V-SCFS19-Super-Quality-Side-Mount-Horizontal-Water-Level-Sensor-Controller-Liquid-Float-Switch-For-Tank/32754969815.html?spm=a2g0v.search0604.3.230.2ca379eeVzPHwB&ws_ab_test=searchweb0_0,searchweb201602_9_10065_10068_319_317_10696_453_10084_454_10083_10618_10307_10301_537_536_10902_10059_10884_10889_10887_321_322_10915_10103_10914_10911_10910,searchweb201603_56,ppcSwitch_0&algo_expid=8cc87956-4194-431d-8e03-3343ebb4f4ef-34&algo_pvid=8cc87956-4194-431d-8e03-3343ebb4f4ef&transAbTest=ae803_5) level sensor in my system.
Can I change the code to work with such a sensor?

vitotai, 2019-02-17:
The code to check "fullness" of water is at

https://github.com/vitotai/BrewManiacEsp8266/blob/37abaa4f665191833882669f77456036f338f21d/src/pins.h#L224

The default code is supposed to work with "Normal Open". If the wort/water level is not high enough to trigger the level sensor, the level sensor is "Open" and the signal should be "High". When the sensor is "triggered", it should be "closed" and short the PIN to GROUND, which results in "LOW" state of the PIN.

If you get opposite result, change the code to

`return (_portvalue & LevelSensorIOExpPin) !=0;`

EroshKA, 2019-02-17:
it worked so well for me

#if EnableLevelSensor
// close/connected/ground: not full
// open/disconnected/V+: full
#if SensorNormalCloseOnNotFull
#if LevelSensorOnIoExpander
bool isWaterLevelFull(void){
	// _portvalue is read for button every loop cycle
	return (_portvalue & LevelSensorIOExpPin) !=1;
}
#else
bool isWaterLevelFull(void){
	return digitalRead(LevelSensorPin) != 1;
}
#endif

#else

#if LevelSensorOnIoExpander
bool isWaterLevelFull(void){
	// _portvalue is read for button every loop cycle
	return (_portvalue & LevelSensorIOExpPin) !=1;
}
#else
bool isWaterLevelFull(void){
	return digitalRead(LevelSensorPin) == 1;
}
#endif

#endif //#if SensorNormalCloseOnNotFull
#endif // EnableLevelSensor

Thanks you