vitotai/BrewManiacEsp8266 #42: Unable to change network

mikkelvp, 2019-03-18:
In web interface. When trying to switch from one wifi network to another, new settings are not applied.
It's in AP mode since the old network is unavailable. Network tab shows that the request gets a 200 response, but it continues to show the old network and stays in AP mode.

Is there other ways to set network settings?

V 0.4.6

vitotai, 2019-03-22:
Try "disconnect" first, then connect the desired network.

mikkelvp, 2019-04-28:
There is no disconnect. Device is in AP mode. Previous connected network is not available.
Old network ssid is "HomeBox...". Setting a new network does not work:
[image]
After new network settings are posted it still remembers the old network:
[image]