vitotai/BrewManiacEsp8266 #43: Multi-Language portugues-br

gumazza, 2019-04-06:
Hi, Vito.

Could you tell me what folder and file I can edit? I want to help you implement the Portuguese-BR language, since I saw that you have improved the OTA update system, giving you the option to choose a language.

arudmin, 2019-04-06:
Take a look through localization folder.

vitotai, 2019-04-09:
Copy english.json from the following folder:
https://github.com/vitotai/BrewManiacEsp8266/tree/master/html.src/src/locales

Rename it to portuguese-br.json, and edit the file.

You might need to setup the environment and modify grunt file.coffee if you want to build by yourself.

Or you can send me the file.

gumazza, 2019-04-10:
Thank you very much friends, I'll work on the files and upload here as soon as they are ready.

Cheers !!!

gumazza, 2019-04-10:
https://github.com/vitotai/BrewManiacEsp8266/tree/master/localization
Are they related to LCD?

If you keep everything within 20 lines, there will be no problems. Right?

vitotai, 2019-04-11:
The localization is for WEB interface only.

Strings on LCD is located in resources.h. You will need to recompile it.

gumazza, 2019-04-11:
That's what I want, translate the LCD of the WEB interface.

vitotai, 2019-04-23:
It is not available but could be done. 
I will make it in next update.

gumazza, 2019-04-29:
Vito, can you provide me the package.json?
It is necessary to install the npm and run the grunt. 
I did not understand how to configure this file.