vitotai/BrewManiacEsp8266 #45: modbus / mqtt intergration

WilliamsonDesign, 2019-05-14:
Is there a possibility of modbus and or mqtt intergration?
Would be great to be able to remotely set to manual , read Temps and Set point from an industrial PLC.
Cheers
Jono

vitotai, 2019-05-14:
Yes. MQTT could be done easily, but I don't know much about modbus or PLC. 
However, I don't think brew "remotely" is a good idea. I am afraid that people really do Brew in a BED and screw up. That is the reason I didn't want to make remote control too easy.

WilliamsonDesign, 2019-05-14:
Ha, who would want to brew remotely and miss out on all the amazing aromas!. When I say "remotely" In my case I am using Brewmaniac in manual as PID's for my HLT and another for MLT and another For the kettle and I would like the PLC (programmable logic controller) to be the central control of all of these instead of the using Brewmaniacs built in auto function.  This is because the PLC I use is already controlling the valves and pump sequencing in the system.

lekrom, 2019-05-14:
Wow. Sounds more complex than it needs to be. Why not just do everything with the PLC? What type of plc is this, just curious...

WilliamsonDesign, 2019-05-14:
AB MICRO850, They need expensive add-ons to read Analog or Temperature sensors. Also this way allows us to control the heat manually, standalone from the PLC system.

vitotai, 2019-05-18:
Full MQTT support, log and control, has been finished on BrewPiLess and waits to be run/verified in a real brew.

MQTT will come to BrewManiacEx after I figure out the use cases.