vitotai/BrewManiacEsp8266 #49: NTC 10K

cherryphilip74, 2019-06-03:
Is it possible to implement the reading temperature also from NTC 10K? The goal is to have two choose, one by 1-wire and the second by A0 pin (3v3,150k,ntc)... There are some All-In One kettle that using NTC probe.
Thanks

vitotai, 2019-06-03:
Multiple 1-wire sensors are supported.

It seems not difficult to support, but the hardware configuration might differs. That says, A0 read VOLTAGE, not resistance. Voltage divider with 3V3/5V is needed. (Well, D1 mini has a voltage divider circuit built-in.) The resistors used will affect the translation of readings and its precision.

Not having the hardware setup, I don't feel like to implement it without the ability to verify.

cherryphilip74, 2019-06-03:
Correct!
Here the proposal
[image]
I tried this project http://esp8266tutorials.blogspot.com/2016/09/esp8266-ntc-temperature-thermistor.html?_sm_au_=iQWdQRHQ23pF5tR6

vitotai, 2023-10-13:
Have you tried the project? How about the readings? 
I don't have experience with NTC 10K sensor, but I've tried A2D with pressure transducer on ESP8266 and ESP32. The reading is not very stable. A2D is also used by WiFi for transmission power, so it works better by turning off WiFi. However, WiFi is the reason for ESP8266/ESP32. I don't see the point to turning off WiFi.

MAX6675 with K-type sensor is used in my coffee roaster project, and it seems to work fine. (However, I had LCD scrambled issues, and the project is suspended.) Therefore, it is easy to support MAX6675. I don't have experience with other sensors.

The problem to support new or other hardware/device is that I don't have the setup. I won't be able to test or verify. I keep maintaining this project because I keep using it. Problems found could then be solved. I would never know what's wrong with other hardware that I don't use.