vitotai/BrewManiacEsp8266 #54: D1 mini board only - howto buttons?

arjepsen, 2019-06-25:
Hey.
I'm trying to get this thing to work, and are working my way around the coding.
I've set:
#define BUTTON_USE_EXT false
in pins.h, since I'm not using the pcf8574, but I'm unsure about how to use the remaining pins for the buttons.
I'm not using a buzzer either, so I'm trying to set it up to use D0, D8, D3 and D4 for the buttons.
The codes compiles and uploads without a problem, but when I short those pins to ground, nothing happens. 
Also, if I go the web interface, and push Manual, and then No, the text start looping fast, so something is running in circles....
Any help?

vitotai, 2019-06-25:
2 buttons configuration is default.
Check 
https://github.com/vitotai/BrewPiLess/wiki/Hardware-Setup
https://github.com/vitotai/BrewPiLess/wiki/Hardware-Setup-example

arjepsen, 2019-06-26:
Does this count for the brewmaniacex code too?
I just tried it again - downloaded the code from scratch, only change is setting #define BUTTON_USE_EXT to false in pins.h
It compiles and uploads fine, and seems to come up working.
However, when I short pin D3 to gnd, nothing happens. When I short D4 to ground, the blue diode lights up - but nothing else. Nothing changes on the lcd.
Do I need to make other changes, since I'm using the wemos D1 mini?

jamesgol, 2019-07-18:
I've been testing this with a wemos d1 mini board and had a similar problem.  It looks like if any of the pins are in a funky state then it causes all of them not to work (the code thinks multiple buttons are pushed ie "fat finger" mode).  Enabling BUTTON_DEBUG helps track this down.  
Using D3, D4, D5, and D6 worked fine for me.  D0 resulted in the cycling behavior, D7 and D8 just caused the "fat finger" failure.   I did not test D1 and D2 since my LCD is hooked there

vitotai, 2019-07-19:
I am sorry for misunderstanding.
Some references:
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html

D8 should be pulled GROUND. You can check the schema of D1 mini.
D7 should work fine. I don't know why it doesn't work for you.
D0 is a special pin which doesn't have internal pull up. If you pull it up externally, it might work. However, I am not sure if that would cause other issue during booting.