vitotai/BrewManiacEsp8266 #55: BrewManiac ESP8266 System update

programinjo, 2019-09-05:
Hi,

I have uploaded firmware to the wemos d1 mini.
Connected to my wifi and open web page of BrewManiac. After that I press "update now" and update stuck. Not movig from [Processing]bm.047.jgz
[image]

vitotai, 2019-09-07:
which browser did you use?
It works for me, so I don’t know what’s wrong.

programinjo, 2019-09-07:
chrome, firefox, edge, internet explorer :)
I tried on two different computers, using two brand new wemos d1 mini freshly opened from the package

vitotai, 2019-09-07:
Try Chrome, and check "Fresh install:Format SPIFFS and download all."
Open Javascript Console and copy the logs. Otherwise, the only thing I can do is saying mine works.

programinjo, 2019-09-07:
>Try Chrome, and check "Fresh install:Format SPIFFS and download all."

when "[Processing]bm.047.jgz"  in console I can see "{finished: 0}"
looks like a loop
[image]
"Updating.." still running

when i refresh the web url after 15 min i get: 
[image]

It happens on two completely different computers. Just to mention that the same hardware work OK with any ESP2866 firmware (for example, both devices are flashed using Fermentrack and registered to the network and working OK) Also, the default password and username has not been changed

programinjo, 2019-09-07:
Just to let you know everything is working fine now.
Now all 3 devices are connected to Unifi equipment and the update works flawlessly :)

For the first time all 3 devices and 2 computers were connected to Mikrotik equipment and the updates did not work. I have no idea what caused this.

If you agree, I will close this issue.

Thank you

vitotai, 2019-09-07:
I am researching this issue but could have found where it goes wrong.
According to the error code (403), the updating script provided wrong information to the controller. 
I would like to see the detail information of the failed request. If it is possible, please click the network request and show me the detail information. Thanks.

[image]

programinjo, 2019-09-07:
Is this enough?

[image]
[image]

vitotai, 2019-09-13:
The update stops because the controller reject the request by returning a "403" error, by which the controller DIDN'T receive enough information. However, the detail information from Chrome showed that required information was sent.

I now have no idea.

tiagojcarlino, 2019-09-13:
Vito my friend, thanks for the returns.
Sorry to take your time, but would you have any idea how to "pop up" the web interface without using the OTA update?
Because I bought the equipment assembled and when I called I came across this problem.
In Update 0.4.6 to 0.4.7 of error: Error open filed!
I have no "tools" to go deeper.
Perhaps a small suggestion is to "delete" the repository files and "upload" them again, with proper permissions.

Hugs

tiagojc!

vitotai, 2019-09-14:
try 
http://bm.local/format-spiffs

to format the file system.

tiagojcarlino, 2019-09-18:
Hello friend Vitto!
It didn't work, I talked to the people I bought the controller from, and eventually sent it to him.
So it has more features and tools for analysis and testing!

Thanks a lot for the help!!

jussipu, 2019-12-11:
I had this same problem and it was in buildflag option missing 1m
`-Tesp8266.flash.4m**1m**.ld`

RonnyRusten, 2020-01-19:
> 
> 
> I had this same problem and it was in buildflag option missing 1m
> -Tesp8266.flash.4m**1m**.ld

This worked for me, seems to be the problem for env:spargeheater and env:dualheater in platformio.ini.

vitotai, 2020-01-22:
Thank you for the information. I thought 4m is equal to 4m1m.(well, I remember that I didn't see a 4m1m script.)

Venturer1804, 2020-03-17:
Just installed your product, and it requires updating. And he himself does not put them. Maybe somehow you can skip them or fix it? And how can I go to a 2-line display?
[image]
Knocks out the error "error: Error open failed!"

vitotai, 2020-03-17:
You must connect to the internet to get the files.

Venturer1804, 2020-03-17:
I go to him through a router, and he, in turn, has access to the Internet. Since I still write to you.

Venturer1804, 2020-03-17:
What data do you need for debugging? Ready to provide

vitotai, 2020-03-17:
You might need to manually upload files.
I had one platform that shares the same issue. I don't have any clue. When the platform connected my computer, everything worked fine. 

The files are at
https://github.com/vitotai/BrewManiacEsp8266/tree/master/data

If you want russian localization, use the bm.htm.gz in this path instead:
https://github.com/vitotai/BrewManiacEsp8266/tree/master/html.src/dist/russian

Venturer1804, 2020-03-17:
I have already tried downloading via http://192.168.4.1:8008/filemenager, but visually nothing happens there. Perhaps you can somehow load them in another way? I have files, I’ve been suffering for 2 days

Venturer1804, 2020-03-17:
I already tried all locations, tried to install older versions, but then it does not load both firmwares. I tried to install all of the latter. By the way, the [Update Note] hyperlink leads to a page that is missing. Maybe updates refer to non-existent places?

vitotai, 2020-03-17:
Would you show a screenshot of the "file manager" with the file uploaded?

allthepies, 2020-03-17:
> I have already tried downloading via http://192.168.4.1:8008/filemenager, but visually nothing happens there. Perhaps you can somehow load them in another way? I have files, I’ve been suffering for 2 days

Your screenshot shows you accessing the UI using the 192.168.0.107 IP address.  You should use this same IP to access file manager e.g.    http://192.168.0.107:8008/filemanager

Venturer1804, 2020-03-17:
[image]
Maybe there are programs to fill the service pack? I have a CP2102 NodeMcu V2 Lua module based on the ESP8266 ESP-12E. Perhaps he needs a different approach?

vitotai, 2020-03-17:
16 files that you have to UPLOAD, by hitting the "Upload" button after select the files.
There is a way. However, I don't even know how to do it under platformio. Here is the information: https://docs.platformio.org/en/latest/platforms/espressif8266.html#uploading-files-to-file-system-spiffs
I've never done that. good luck.

Venturer1804, 2020-03-17:
> Your screenshot shows you accessing the UI using the 192.168.0.107 IP address. You should use this same IP to access file manager e.g. http://192.168.0.107:8008/filemanager

Thank you for trying to help. But I already did it too

Venturer1804, 2020-03-17:
ESP already saves files, but when updating it returns "error: connection refused"

[image]
[image] 

Where to dig?

Venturer1804, 2020-03-17:
I played around with the bootloader settings, it turned out that I have a flash size of 16M-C1. Although the developer indicated 4M. But with these settings, I managed to manually fill in the update files. Thanks to everyone who did not remain indifferent. Thanks to the developer for the sketch and the time taken! Hope my solution helps someone too

wagnerlibardi, 2021-05-12:
> Hello!
> I am also having the same problem, to upgrade BM 0.4.6 to 0.4.7.
> I'm from Brazil.

Hi Thiago

I need help. I'm from Brazil too.
What program are you using to compile and write? Some errors are appearing in my process

Thanks

Wagner Biagioni Libardi
wagner.libardi@hotmail.com