vitotai/BrewManiacEsp8266 #59: Non latin characters break web LCD

dodther, 2019-12-09:
[image]
[image]

vitotai, 2019-12-10:
Obviously, that character is not in the FONT library of Javascript. Does your LCD show the character?

dodther, 2019-12-10:
My LCD show all characters.

vitotai, 2019-12-10:
[bm.english.htm.gz](https://github.com/vitotai/BrewManiacEsp8266/files/3945410/bm.english.htm.gz)

If you are using 0.4.8 HTML/Javascript, you might try this file by uploading as "bm.htm.gz".

dodther, 2019-12-11:
File upload didn’t help.

vitotai, 2019-12-12:
Did you force reloading the page? Although the file I provided doesn't contains cyrillic font but European font, you should see something else.
I don't have the font data of cyrillic right now. A way to get font data must be developed unless the font library is provided.

dodther, 2019-12-13:
I reloading the page. Can I help with fonts?

vitotai, 2019-12-13:
[bm.htm.gz](https://github.com/vitotai/BrewManiacEsp8266/files/3960717/bm.htm.gz)

Sorry, I think I gave a wrong file. Try this one with cyrillic fonts.