vitotai/BrewManiacEsp8266 #60: OLED instead of LCD

Karteesi, 2020-01-11:
I just started building a BrewManiac. The only thing I have working now is the 20x4 LCD with the 44780 circuit.  As it did not work right away (see below, *), I looked at the source code and noticed that there is support for an OLED display, which I would like better. However, the display that I have is of type SH1106, and the BrewManiac code is for type SSD1306. Is the difference critical?

*There is a #define in pins.h like **#define PCF8574_ADDRESS 0x20**. However, when the LCD is set up in ui.h, it is done like this: **LiquidCrystal_I2C lcd(0x27, 20,4,YWROBOT);** Shouldn't that be **LiquidCrystal_I2C lcd(PCF8574_ADDRESS, 20,4,YWROBOT);**?

vitotai, 2020-01-12:
The #define's doesn't mater, but PCF8574_ADDRESS maters. Check uiScanLcdAddress(), and you will find that the address of LCD is automatically detected.

SH1106 should work as SSD1306 if you change the "SSD1306" to SSH1106 at line26 in ui.h. However, I haven't tested OLED code for a while and have not confidence that it would work fine.

Karteesi, 2020-01-12:
> The #define's doesn't mater, but PCF8574_ADDRESS maters. Check uiScanLcdAddress(), and you will find that the address of LCD is automatically detected.

OK; I checked uiScanLcdAddress()  and the problem seems to be that in the original circuit, there are **two** PCF8574's. One for the buttons and one for the LCD. I only have one for now, that is, one for the LCD.

Its address is 0x20, which the scan in uiScanLcdAddress() skips, because in the original circuit that is the address of the buttons.