vitotai/BrewManiacEsp8266 #61: Buttons  without (second) PCF8574

Karteesi, 2020-01-12:
I built a simple 4-button "keyboard" and had the idea of connecting it directly to the NodeMCU board. That is, without the intervening PCF8574. There seem to be just enough available pins: D3, D4, D5, D9, and D10. Even if a pump takes up D5, 4 pins for 4 buttons should still be left.

Can you give any pointers to the source code for reading the buttons? I **think** it should be rather simple to get the values - but I'm not sure...

vitotai, 2020-01-13:
pins.h

Karteesi, 2020-01-13:
Thank you! Awesome – the code was already there!

vitotai, 2020-01-13:
I forget to mention that I've tried TX/RX and failed for some unknown reason. If you failed to use them, I won't be able to help.

Karteesi, 2020-01-14:
I failed at first. but then I realized they were used by the USB connection. Once I unplugged the USB, D9 was good for the ENTER button. (Of course, I then had to add a separate PSU...)