vitotai/BrewManiacEsp8266 #64: Web ui symbols

RonnyRusten, 2020-01-21:
[image]
* The symbol for Sparge heater is missing. It shows correct on LCD (the letter S inverted), but on web lcd it says 1.
* Degree Celsius symbol is "funky" on set value. The same error is on LCD.

allthepies, 2020-01-22:
Additionally the spelling mistake on the panel title still exists "Cotnroller" should be "Controller"

vitotai, 2020-01-23:
OK. I know this issue, but it would take some time to fix, excluding the "controller"one.

RonnyRusten, 2020-01-23:
I have replaced all occurrences of "Cotnroller" that VS Code could find, but it still shows "Cotnroller".
If you could tell me where in code the error might be, I can look into it... also the "funky" symbols...

RonnyRusten, 2020-01-23:
Sorry, I'm an idiot. 
The symbol behind 34.75 isn't the deg celcius, it's the SetPoint symbol...
The symbol under the Heat symbol when both Heat and Sparge Heat is activated is just an regular letter S, I guess, therefore it isn't in the definitions. If you turn off heat, it switches to a 1, which is defined as primary heat, which I believe is wrong, it should either be 2, or the inverse S, also, the inverse H should turn into a regular H.

vitotai, 2020-02-04:
I thought I didn't consider some scenarios, but it turns out I did. This bug is solved by correcting the glyphs in Javascript. Try download bm.htm.gz from html.src/dist/english/ and try again. Don't forget to reload the page.

RonnyRusten, 2020-02-06:
Yes, this worked for the symbols, but it still says "Cotnroller". I tried to change it in bm.htm.gz and upload again, but it didn't work. Is this defined somewhere else?

vitotai, 2020-02-07:
Well. modifying the bm.htm.gz should fix it. You might need to reload again.

RonnyRusten, 2020-02-07:
I finally got it to work, but I had to open the bm.htm.gz and change it directly in 7zip editor. How do I build a new bm.htm.gz? I have tried to change some of the other texts, but I can't figure out how this works. And, where does the other texts come from? I have changed texts in english.json, but nothing is changing on the web, and the date/time on bm.htm.gz isn't updated when I run the Build task.

vitotai, 2020-02-10:
The building process of HTML/JS files is defined in grunt.coffee.