vitotai/BrewManiacEsp8266 #69: Temprature does not update

KLelong, 2020-06-14:
Hi,
I'm modifying BMex 0.4.9 to suit my hardware. It seems to work, but the temperature display is not updated. The initial reading is there, but when I warm up the sensor, nothing happens.
When I try a manual run, the same happens : no temperature updates.
I don't have a i2c display, nor i2c buttons. I used a ssd1360spi display and I got it working. I also changed the heater and pump pins. The i2c pins are unused !
I use the web-interface to control the device.
I tried debugging this, but I'm stuck. I see it regularly goes to tpReadTemperature(), but it never passes tpSensorDataReady().
I made an Arduino sketch using the same code (temperature part) 
[ds1820-read.tar.gz](https://github.com/vitotai/BrewManiacEsp8266/files/4776791/ds1820-read.tar.gz)
and there the temperature updates fine.
Any suggestions ?
TIA

vitotai, 2020-06-15:
Check your hardware.

KLelong, 2020-06-15:
Thanks for the reply.
What hardware do I need to check ?
Like I said, the Arduino sketch runs. That's on the same Wemos-module, nothing changed concerning hardware.
Is the reading of the temperature depending on other hardware to work ? If so, what ?
TIA