vitotai/BrewManiacEsp8266 #70: Exception and reboot loop

allthepies, 2020-06-23:
Just built and uploaded the latest version to a NodeMCUv2 and also a brand-new WemosD1 Pro.
Both units throw an exception and reboot as soon as the web interface connects.

`dhcp client start...
ip:192.168.1.17,mask:255.255.255.0,gw:192.168.1.1
local IP: 192.168.1.17ws[/ws][1] connect
ws[/ws][1] pong[0]:
Fatal exception 3(LoadStoreErrorCause):
epc1=0x40100dd1, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40057368, depc=0x00000000

Exception (3):
epc1=0x40100dd1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40057368 depc=0x00000000

>>>stack>>>
`
Using these build flags

`platform = espressif8266@~2.3.3
board = nodemcuv2
framework = arduino
build_flags = -Wl,-Tesp8266.flash.4m2m.ld
    -DPUMP_INVERTED_LOGIC=true
    -DSupportDistilling=false
     -DSpargeHeaterSupportEnabled=false
   -DDebugOutput=true
`

Any ideas ?

vitotai, 2020-06-24:
No, but let me share my experience.

About two weeks ago, I was developing the 0.5 version of BrewManiacEx, which has been for a while. The test platform ran well until someday when everything went wrong. I accidentally drop a few water around the breadboard, but I was  sure that the water didn't touch the breadboard. However, unexpected exceptions happened since that, and I had no idea what caused that. That exception happened at almost the same action. The mystery is that even I commented out the code for responding the action the exception still happened.  I rolled back and tested old working versions, but the exception still happened. It took me about three days before I gave up to change to another D1 mini. Everything works fine, and the exception doesn't happen on the D1 mini.
I have totally no idea what happened. The triggering point is sending HTTP response, but other calls to send HTTP response didn't trigger exceptions, which made me believe that it's software issue. And, I "solved" it by change hardware?

allthepies, 2020-06-24:
OK, but for me the 0.49 version, either pre-built or compiled myself doesn't work on two different ESP devices.  Both show the same failure i.e. the exception and reboot as soon as the Web UI connects it's websocket.  Is there an "easy" way I can get back to the 0.47 code to compile myself ?

vitotai, 2020-06-24:
https://github.com/vitotai/BrewManiacEsp8266/releases

allthepies, 2020-06-24:
> https://github.com/vitotai/BrewManiacEsp8266/releases

Unfortunately, doesn't work :(

allthepies, 2020-06-24:
Found the issue, my Wifi SSID name is long and blows a char[128] buffer, which causes BMEx to crash.
Pull request created with the fix
https://github.com/vitotai/BrewManiacEsp8266/pull/71

vitotai, 2020-06-25:
Good job. However, the maximum SSID length is 32 characters according to the standard.

allthepies, 2020-06-25:
> Good job. However, the maximum SSID length is 32 characters according to the standard.

Yes, that's right.  But for this buffer, the SSID is only part of a longer string which is being written into the buffer.  If you SSID starts to approach 32 chars (as mine does) then it pushes the overall length of the string to over 128 bytes.

	char buf[128];
	
	sprintf(buf,"{\"host\":\"%s\",\"secured\":%d,\"wifi\":%s}",_gHostname,_gSecuredAccess? 1:0,WiFiSetup.status().c_str());

This one needs to be fixed IMO.

closed 2020-07-05 by allthepies