vitotai/BrewManiacEsp8266 #78: Integration with BrewFather

WojtekGD, 2021-01-12:
Hallo  Vito
Is there an option to cooperate and connect Brewmaniac with the BrewFather program?

https://brewfather.app/

vitotai, 2021-01-12:
how, what?

WojtekGD, 2021-01-12:
Now BrewFather works with SmartPID and test with GrainFather controller. Control of mashing and cooking from the BrewFather level on the Brewmaniac would be a very nice solution.

planbnet, 2021-05-05:
Brewfather already integrates BrewPiLess: https://docs.brewfather.app/integrations/brewpiless

Implementation of this probably possible just from their side, the necessary APIs should be available.

vitotai, 2021-05-14:
I've contacted Brewfather for a generic MQTT protocol. The SmartPID implementation is proprietary and only for SmartPID.

shingle01, 2022-03-19:
Hey @vitotai, did you get any traction with Thomas from Brewfather?  I doubt he will support MQTT but you never know. After watching a recent Kegland webinar it sounds like Rapt integration with brewfather is imminent which may include Brewzilla gen 4 support.  It sounds like webhooks are Thomas' preference.  Perhaps webhook polling to facilitate BM control?