vitotai/BrewManiacEsp8266 #79: Modifying web-interface (beta 0.5)

KLelong, 2021-01-12:
Hi,
For my installation I need to modify the settings : pidstart should be greater than 3.5. 
I modified max in "s_pidstart" in setting_resources.js, recompiled and uploaded, but the new value does not get included.
What am I doing wrong ?
TIA

vitotai, 2021-01-13:
reload? maybe.

KLelong, 2021-01-13:
Thanks, but reload does not help, tried several times.
I modified the code in two places : that s_pidstart and in BrewManiac.cpp in the heatLoadParameters function. I think there are still other places.
I'll have to find some real buttons and see what happens then. At the moment I'm doing all via the web-interface.

KLelong, 2021-01-13:
I realised a jumper-wire will do as button. Tried and the same problem is there.
Searching further, I stumbled on the file 5.WebInterface.md stating that html/javascript files will be uploaded. Was that the "reload" you mentioned ? If so, how do I do that ? Using the filemanager ?
TIA

vitotai, 2021-01-14:
"Reload" the page. Sometimes, you might need to click the reload button to force the browser to really reload the page.

KLelong, 2021-01-22:
Hi,
Actually, my question was : how to integrate the modified html-source-code in the firmware. I.e. I modified the html-source-code, but then it needs to be transferred to data_edit_html_gz.h (I think). How do I do that ? Is there a platformio-tool ? Or what should I use ?
TIA

vitotai, 2021-01-22:
grunt tool to "build", and "gene.sh" in the source to convert to C header file.

KLelong, 2021-01-22:
For others :
I went to the html-directory.
I had to install npm, then 
```
sudo npm install -g grunt 
npm install 
npm install --dev coffeescript 
grunt build
```
But then ? You said gene.sh in the source directory, but I can't find it anywhere.
P.S. I only recreated my steps afterwards, so maybe I missed something.

vitotai, 2021-01-23:
it's a simple shell script.
I forgot to add to git.

`#!/bin/sh

OUTDIR="cheader"
DISTDIR="dist"

if [ ! -d $OUTDIR ]; then
    echo "$OUTDIR not found!"
    mkdir "$OUTDIR"
fi
rm $OUTDIR/*.h

htmlfiles=(bm.htm.gz)

variables=(data_bm_htm_gz)

outfiles=(bm_htm)

languages=(english spanish portuguese-br russian italiano)

gen_C_file()
{
lang=$1
for ((index=0; index<${#htmlfiles[@]}; index++)); do
    srcdir="dist/$lang"

#   echo "[$index]: ${htmlfiles[$index]}"
   input="$srcdir/${htmlfiles[$index]}"
   output="$OUTDIR/${lang}_${outfiles[$index]}.h"
   variable=${variables[$index]}
   #echo "input: $input output file: $output with variables $variable "
   xxd -i  "$input" > $output 
   echo "processing $output"
   sed -i "s/unsigned char .\+\[\]/const unsigned char $variable\[\] PROGMEM/" $output
done
}

for lang in "${languages[@]}"
do
gen_C_file $lang
done`

lekrom, 2021-03-10:
I was wondering how you converted the embedded web pages. Clever...

lekrom, 2021-05-03:
Hi Vito,

I tried building the C header without success. I can not get this script to work. I get this error:
$ ./gene.sh
rm: cannot remove 'cheader/*.h': No such file or directory
./gene.sh: 18: Syntax error: "(" unexpected

Please advise.