vitotai/BrewManiacEsp8266 #83: MQTT For Node Red Integration

interstaller555, 2021-02-28:
is MQTT integration available for brew maniac? would want to be integrated with node red.

vitotai, 2021-03-01:
Not yet. I don't have use cases. that is, I don't know how users are going to use it.

interstaller555, 2021-03-01:
Is there a way I can reach out directly? I have some pretty compelling use cases.

vitotai, 2021-03-01:
You can attach files, here.

interstaller555, 2021-03-01:
The reason I would like to see node red/mqtt integrations is as follow: this is a scenario for a 10bbl brewery (mine) as well as my pilot brewery (1bbl)

System requirements:
Wemos D1 mini
LCD 20x4
3 Input (max31865/DS-18b20/DHT22/Hall effect/Pressure)
3 Output (relay)
	Control Select - PWM, Hysteresis (On/off),PID 
	Hysteresis - support DHT, Pressure, or HE
	Pump
Local Control (4 push Button and or Rotary Encoder)
	-Rotary Encoder reduces IO requirements

Layout:
2 sections
1- Pilot brewery (applies to 100% home-brew setups)
-=Current Setup=-
1bbl HLT (5.5kw) Single Element (Would like to upgrade to two at some point soon)
20G Mash Tun (SSBT InfuSSion)
	RIMS (3.4Kw)
	24VDC Heating Pad (under Tank for “Long term souring”)
1BBL BK (6.5kw) Sinlge Element - soon to be two)

Proposed Integration (1 Wemos Per Kettle)
 	- HLT
		Inputs:
		Temp
		High Water Level Sensor (Float Sensor)
		Low Level Sensor (Element Burnout Prevention)
		Hall effect Sensor (Fill level)
		Outputs:
		Heating Element (SSR)
		Fill Valve
		Temp Set Point Indicator (“You’re ready to dough in”)

	- MLT
		Inputs:
		Mash Core Temp
		RIMS/HERMS Outlet Temp
		Outputs:
		RIMS (SSR)
		Heating Pad (SSR)
	-BK
		Inputs:
		Hall effect - fill volume
		Low Level/High Level Sensors (Or can be done with HE Sensor)
		Outputs:
		Heating Element (SSR)
		
Node Red Integration
Flows control all components independently but display a unified dash.
Kettle Type (HT, MLT, BK, Fermenter, Brite)
Control Select: ie. HLT-PID, RIMS - PID, BK- PWM, Fermenter - Hysteresis
Pump Control
Recipe integration (brew father, or other)
Data logging
Front end interface.
**Wemos retains 20x4 as well as all other local control functions. Can be set to control remote or locally
Modularity allows for growth without losing current hardware.
Unique ID per wemos that is tied to each controller. - important in the cellar

Cellar 
Home-brew or professional setup

-Pressure, Temp, control (Cooling/Heating)
Tilt integration - important for spunding or fermentation hang

Automated spunding and or crash/carb on input (tilt/ispindel or time set point)

Carbonation control - set #volumes and temp - crash to temp *THEN* open valve for carb stone.

Retain said carbonation based on head pressure etc.

I can go into the feather valves and how I would love to be able to integrate this into my steam fired system but that just seems like too far in the future.

For Nodered via MQTT - it seems like the legwork to incorporate MQTT and allow the control is *relatively easy* it would also allow users to customize their hardware to suit their exact needs. Furthermore. - the cost of a few demos D1’s and being able to switch them out is significantly lower than say - losing a batch of beer because the singular memos dies or burned up or - whatever...

As a professional brewer this is where I would love to see this type of integration - as a home brewer (of 18 years) this is the kind of hands on integration that makes most sense.

/s

vitotai, 2021-03-03:
1. What you described is very different from BrewManiacEx.
Most of the jobs done on BrewManiacEx are for automation and user interface, including Web pages. What you need is a generic wireless controller that can connect devices/sensors. CraftbeerPi might be a better choice, though I haven't spent time studying it. 

2. I would suggest ESP32 instead of ESP8266(D1 mini). ESP32 has far more pins and runs faster. It doesn't cost too much more.

3. I prefer Rotary Encoders over buttons because only one hole is needed on the project box. However, from hardware and software aspect, buttons are better. Yes, simply better. Rotary encoder needs 3 pins, RA/RB and Push, while 3 buttons can be used as exact function: UP/DOWN and Enter. Long pressed can even be implemented for buttons.Rotary encoder works best with interrupt while polling is fine with buttons.

4. The fermentation control is implemented as BrewPiLess, my other project. All functions except carbonation valve you mentioned are supported. ESP32 is required to support TILT, though. 

BrewManiacEx is designed for single vessel E-BIAB systems. I know some people had used it in 3-vessel systems (maybe) because it is more flexible than regular PIDs, but I wouldn't recommend it. If I am going to develop a 3-vessel system, I would design a generic controller like what you describe. Because I am using an E-BIAB system, so I keep working on current BrewManiacEx.

interstaller555, 2021-03-03:
Cbpi isn’t robust enough IMO and relies solely on a pi. The automation portions of brew maniac are what I’m after - the integration with node is specifically for front end control. I have esp32 as well as d1’s - the truth is there’s not a whole lot of io requirements.I think adding mqtt support to your current software setup would do exactly what I’m after and provide the expansion abilities.

vitotai, 2021-03-11:
The problem I have is that I don't have exact use cases of MQTT. To be exact, what states/signals/messages are users interested in. Obviously PUMP and TEMP are two most important ones, but how about others, like states of automation, hop schedule?
It would be simple if MQTT is only supported under manual mode.

interstaller555, 2021-03-11:
I think the whole point would be to have mqtt jjst for a “manual” mode so you can create your own automation flows in node red

interstaller555, 2021-03-11:
Basically being able to have an address ID for every io with mqtt would be (IMO) the easiest way to integrate.