vitotai/BrewManiacEsp8266 #91: Another Sparge error

lekrom, 2021-06-18:
Hi @vitotai 
I just upgraded to 0.5.
While doing a test run, I noticed an issue with the sparge heating control:
After reaching temp for mash step 1, the sparge heater kicked in as expected. When I cancelled the process and resumed it, the sparge heater stayed off. I specifically tested this as the prev version sometimes had the sparge heater on permanently (even boiling the sparge kettle) when pausing and resuming.

lekrom, 2021-06-23:
I noticed that the Sparge heating only gets initiated in the "Doughing in" stage ie. in the autoModeEnterDoughIn() function, line 4405-4410 in Brewmaniac.cpp:
```
#if SpargeHeaterSupport == true
	if(gEnableSpargeWaterHeatingControl)
	{
		startHeatingSpargeWater();
	}
#endif
```
Thus, if the auto process is resumed after this point, the sparge heater will never be initialised.

Did I miss something?

vitotai, 2021-06-24:
You are right. It's missed.
There are too many functions that I don't use.

lekrom, 2021-06-24:
Since sparge can be disabled for a brew by selecting "no" when asked "sparge water added", the brewlogger will have to keep track of this choice as well, to avoid starting heating an empty sparge vessel when resuming.