vitotai/BrewManiacEsp8266 #92: Network issue

lekrom, 2021-06-18:
I am unable to connect BM to my wifi network. After connecting, the sys reboots and displays the last 5 digits of its IP adress. (Nothing else on the LCD).
I managed to briefly connect to it via my wifi but it hangs and reboots before the page is done loading.

In AP mode only, the web interface works perfectly.

lekrom, 2021-06-26:
I hooked up the serial monitor to see investigate issue, this is what it spits out (SSID and Passw redacted):
`File System begin() Success
read 166 bytes:{"host":"bm","user":"brewmaniac","pass":"rdwhahb!","secured":0,"ip":"0.0.0.0","gw":"0.0.0.0","nm":"0.0.0.0","ap":0,"ssid":"XXXXX","stapass":"XXXXXX"}
hostname:bm, user:brewmaniac, pass:rdwhahb!, secured:0
load:"{"rest_tm":[1,30,60,30,0,0,0,10],"rest_tp":[41.9,40,60,70,0,0,0,75],"boil":60,"hops":[60,10]}"
NO HS session
loadSensorSetting
Sensor: addr:112, 0x28FF62D16116047D
Sensor: addr:120, 0x28FF6BDC02170315
Sensor: addr:128, 0xFF00000000000000
invalid sensor address! 6Number of sensors:2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with CASADIMORRI [2Ghz], channel 8
dhcp client start...
ip:192.168.101.110,mask:255.255.255.0,gw:192.168.101.1
delay
gButtonPressed=0
gButtonPressed=0
gButtonPressed=0
begin:: 3 ,SSID:XXXXXX pass:XXXXXXX IP:192.168.101.110, gw:192.168.101.1

AP_STA mode
scandone

create network:bm pass:rdwhahb!
Connected! IP address: 192.168.101.110HTTP server started
HTTP Update server started

Set sensor for 0, P:0, Aux:1
local IP: 192.168.101.110End Setup
Sensor disconneced!
Sensor disconneced!
Sensor disconneced!
Sensor disconneced!
Sensor disconneced!
Sensor disconneced!
Sensor disconneced!`

lekrom, 2021-06-26:
In AP only mode, all works fine. So not a real sensor issue as reported.

lekrom, 2021-07-02:
Hi @vitotai 
Its seems to go wrong at the last step in the setup() routine, Line 1590 in BrewManiacex.cpp
`displayIP(WiFiSetup.isApMode());`
If I comment this line out, all works as expected.

vitotai, 2021-07-03:
I don't understand how could it be.
That function just displays IP on LCD.

Malord958, 2022-04-06:
I have had this issue in version 0.5.1 only with 12-digit IP addresses (for example 192.168.123.123)
with an IP address of 11 digits (192.168.123.12) there is no problem.