vitotai/BrewManiacEsp8266 #93: PID autotune does not work correctly

lekrom, 2021-06-18:
The PID autotune algorithm behaves erattically.
The autotune cycle completes prematurely without cycling through the setpoint as it should. The PID params derived are nonsensical and unrepeatable. Not working at all.

felipeserta, 2021-06-24:
Depending on phisical behavior of the system, heatting power isolation etc, the autotune get lost detecting the cycles, this is first problem, second values calculated in general (my system with 5kw heater and kettles isolated that means long cycles) are over memory capacity. Is possible to calculate by hand and is better to reduce noise value to 0.1 or less, getting max and min values excluding first 2 cycles. I changed to another library much better 
// PID automated tuning (Ziegler-Nichols/relay method) for Arduino and compatible boards
// Copyright (c) 2016-2020 jackw01
// This code is distrubuted under the MIT License, see LICENSE for details
The actual lib get lost maybe configuring better lookback paramaters migth solve the fisrt issue. The second must be change kp ki and kd to word, not simple but possible.

lekrom, 2021-06-24:
Hi @felipeserta,

Thanks for the info, I will have a look at this autotune lib. 

I am well aware of Ziegler Nichols and PID control in general. Have done quite a lot of it in my life. 
The autotune worked (reasonably) fine on my system on previous versions though. Previously, whilst it was cycling I recorded the peaks and times and did my own Z-N calc as a sanity check and got fairly similar tuning parameters. 

I will take a deep dive into it this weekend to figure out where it goes wrong. At first glance, I see nothing to explain why this version is misbehaving in this way. The thing is, from my observations, the autotune algorithm did not behave as expected, it cycled only once, it should cycle at least 3 times. It just occurred to me that I may have used a lower sensor resolution previously but I cant remember exactly, this will also implicitly affect the noise level. Something to experiment with at least.

With regards to my system, I have a 70l kettle (mostly doing 40-50l batches) with a 4.5kW element, no insulation at the moment. Oscillation period is typically around 6 minutes. 
I did  manage to manually dial the parameters in pretty well though.

felipeserta, 2021-06-24:
@lekrom nice! The kind of test i did at the past was adding some serial.prints at the code monitoring max min and tu(period),cycle etc, some pidautotunelookback parameters and pidautotunenoise changes also. I use arduino ide to monitor because VS code had some limitation deppending of license. Simulate by holding the main sensor and let go when heat is of (relay method ) using smal setpoint 31C for instance. Looking at the trends judge if its ok. Good luck