vitotai/BrewManiacEsp8266 #94: heat control heatWindowSize

felipeserta, 2021-06-22:
Nice new interface 0.5, how to update de htm file by hand? It does not appear at bm.local:8008/filemanager I changed calibration functiion in the code but I dont know how to change de htm now.

Small issue at ln 2107 that keep blinking primary heater SSR or relay once per cycle at max output 255.

  	if ((pidOutput / 255) * ((unsigned long)_heatWindowSize * 250) > now - _windowStartTime)

corrected
  	if ((pidOutput / 255) * ((unsigned long)_heatWindowSize * 250) >= now - _windowStartTime)

Allways amazing work at BMEx

felipeserta, 2021-06-22:
It does not appear at bm.local:8008/filemanager I changed calibration functiion in the code but I dont know how to change de htm now. Im using esp32

lekrom, 2021-06-23:
@felipeserta,

The html files are embedded in the code, this process is explained in issue #79. #

felipeserta, 2021-06-23:
@lekrom thanks! How did you run the script? Im using VScode +platformio and have no ideia. As commented in issue 79 looks like something in linux.

lekrom, 2021-06-23:
Jip,
It is a shell script. I ran it linux (Ubuntu) (had to change the shebang from #!/bin/sh to #!/bin/bash to get it to work).
I believe the script will also run in IOS. On a linux system you can run the script from the normal terminal or the Platformio terminal, both works.
I read somewhere that you can run bash scripts in windows but need to install cygwin. I have not tried this though as I do all my coding on my linux machine.

felipeserta, 2021-06-23:
@lekrom Thanks a lot. I´m using Windows só i will look how to run those scripts. I think the web page in the code works better but I can paste the web page in spiffs and works fine also.

lekrom, 2021-06-24:
I dont believe the webpage in spiffs is supported anymore.