vitotai/BrewManiacEsp8266 #98: Size of WiFi password

KLelong, 2021-07-25:
Hi,
I was working on new hardware, so I purchased a new ESP32-device and flashed it, with the beta-5 software.
Unfortunately, I can't persistently connect to my WiFi with that device. I found that passwords bigger than 40 characters are not saved.
When I connect to the bm-AP, I can scan and connect to my own AP, but when I reboot, the connection is not restored. I looked at the serial debug-log and this is what I see when the password is saved : 
```
processRemoteCommand:"{"nw":"<SSID>","pass":"<AP-password>,"wificmd":"con"}"
Connect to <SSID> pass:<AP-password>
sav netcfg:{"host":"bm","user":"brewmaniac","pass":"rdwhahb!","secured":0,"ip":"0.0.0.0","gw":"0.0.0.0","nm":"0.0.0.0","ap":0,"ssid":"<SSID>","stapass":null}Change Connect
**try::SSID: pass:<AP-password> IP:0.0.0.0, gw:0.0.0.0
local IP: 192.168.89.242BmwHandler canHandle file:/canonical.html
```
SSID and AP-password are modified for privacy. AP-password is more than 40 characters long.
The local IP is in the right subnet, so the connection works, but the saving does not work. I can connect to that ip-address to get the web-site.
But after reboot : 
```
....
Start..
File System begin() Success
read 147 bytes:{"host":"bm","user":"brewmaniac","pass":"rdwhahb!","secured":0,"ip":"0.0.0.0","gw":"0.0.0.0","nm":"0.0.0.0","ap":0,"ssid":"<SSID>","stapass":null}

```
I searched the code for a location where that password is checked and/or stored but I can't find anything.
The strange thing is that with my previous ESP32-device I could connect with the same software. Unfortunately I can't remember how I did this. I reflashed that device and it did still work. But when I saved a smaller password and again with the real password it also didn't save that password.

Any help ?
TIA

KLelong, 2021-07-25:
I found an old network.cfg and I uploaded that. Now my ESP32 connects fine.
In that file the whole wifi-password is recorded.

vitotai, 2021-07-27:
I didn't check the specification but per the result of googling, maximum pass phrase length is 32.
The storage of password is 32 only.

KLelong, 2021-07-31:
I tried to search the sources, but there the WPA2 spec is followed : password-length=64 bytes.
I found this in `.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp32/esp_wifi_types.h` :  
```
/** @brief Soft-AP configuration settings for the ESP32 */
typedef struct {
    uint8_t ssid[32];           /**< SSID of ESP32 soft-AP */
    uint8_t password[64];       /**< Password of ESP32 soft-AP */
    uint8_t ssid_len;           /**< Length of SSID. If softap_config.ssid_len==0, check the SSID until there is a termination character; otherwise, set the SSID length according to softap_config.ssid_len. */
    uint8_t channel;            /**< Channel of ESP32 soft-AP */
    wifi_auth_mode_t authmode;  /**< Auth mode of ESP32 soft-AP. Do not support AUTH_WEP in soft-AP mode */
    uint8_t ssid_hidden;        /**< Broadcast SSID or not, default 0, broadcast the SSID */
    uint8_t max_connection;     /**< Max number of stations allowed to connect in, default 4, max 10 */
    uint16_t beacon_interval;   /**< Beacon interval which should be multiples of 100. Unit: TU(time unit, 1 TU = 1024 us). Range: 100 ~ 60000. Default value: 100 */
} wifi_ap_config_t;

/** @brief STA configuration settings for the ESP32 */
typedef struct {
    uint8_t ssid[32];      /**< SSID of target AP*/
    uint8_t password[64];  /**< password of target AP*/
    wifi_scan_method_t scan_method;    /**< do all channel scan or fast scan */
    bool bssid_set;        /**< whether set MAC address of target AP or not. Generally, station_config.bssid_set needs to be 0; and it needs to be 1 only when users need to check the MAC address of the AP.*/
    uint8_t bssid[6];     /**< MAC address of target AP*/
    uint8_t channel;       /**< channel of target AP. Set to 1~13 to scan starting from the specified channel before connecting to AP. If the channel of AP is unknown, set it to 0.*/
    uint16_t listen_interval;   /**< Listen interval for ESP32 station to receive beacon when WIFI_PS_MAX_MODEM is set. Units: AP beacon intervals. Defaults to 3 if set to 0. */
    wifi_sort_method_t sort_method;    /**< sort the connect AP in the list by rssi or security mode */
    wifi_scan_threshold_t  threshold;     /**< When scan_method is set, only APs which have an auth mode that is more secure than the selected auth mode and a signal stronger than the minimum RSSI will be used. */
} wifi_sta_config_t;

```
Unfortunately, I can't find if that code is actually used by the software.

vitotai, 2021-08-22:
It's defined in BPLSettings.h.
If you change the structure, all old settings are gone.