vitotai/BrewManiacEsp8266 #102: Compile error when LCD_USE_SSD1306 set to true

shingle01, 2022-03-07:
I thought I'd see if I could get the OLED display working on esp32.  Unfortunately, I get the following compile error.

Compiling .pio\build\esp32\src\FsEeprom.cpp.o
In file included from src/BrewManiac.cpp:354:0:
src/ui.h: In function 'void uiLcdDrawSymbol(byte, byte, byte)':
src/ui.h:264:96: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
   display.drawXbm(x,y,FontWidth,FontHeight,(const char*)SymbolMaps[_reservedChar2SymbolId[sid]]);
                                                                                                ^
In file included from lib/esp8266-oled-ssd1306-4.0.0/src/SSD1306Wire.h:34:0,
                 from lib/esp8266-oled-ssd1306-4.0.0/src/SSD1306.h:33,
                 from src/ui.h:24,
                 from src/BrewManiac.cpp:354:
lib/esp8266-oled-ssd1306-4.0.0/src/OLEDDisplay.h:175:10: note:   initializing argument 5 of 'void OLEDDisplay::drawXbm(int16_t, int16_t, int16_t, int16_t, const uint8_t*)'
     void drawXbm(int16_t x, int16_t y, int16_t width, int16_t height, const uint8_t *xbm);
          ^
In file included from src/BrewManiac.cpp:354:0:
src/ui.h: In function 'void uiLcdDrawSymbolBmp(byte, byte, SymbolId)':
src/ui.h:282:94: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
  display.drawXbm(XofCol(col),YofRow(row),FontWidth,FontHeight,(const char*)SymbolMaps[symbol]);
                                                                                              ^
In file included from lib/esp8266-oled-ssd1306-4.0.0/src/SSD1306Wire.h:34:0,
                 from lib/esp8266-oled-ssd1306-4.0.0/src/SSD1306.h:33,
                 from src/ui.h:24,
                 from src/BrewManiac.cpp:354:
lib/esp8266-oled-ssd1306-4.0.0/src/OLEDDisplay.h:175:10: note:   initializing argument 5 of 'void OLEDDisplay::drawXbm(int16_t, int16_t, int16_t, int16_t, const uint8_t*)'
     void drawXbm(int16_t x, int16_t y, int16_t width, int16_t height, const uint8_t *xbm);
          ^
In file included from src/BrewManiac.cpp:354:0:
src/ui.h: In function 'void uiLcdInitialize()':
src/ui.h:299:31: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
     display.setFont(Cousine_10);
                               ^
In file included from lib/esp8266-oled-ssd1306-4.0.0/src/SSD1306Wire.h:34:0,
                 from lib/esp8266-oled-ssd1306-4.0.0/src/SSD1306.h:33,
                 from src/ui.h:24,
                 from src/BrewManiac.cpp:354:
lib/esp8266-oled-ssd1306-4.0.0/src/OLEDDisplay.h:201:10: note:   initializing argument 1 of 'void OLEDDisplay::setFont(const uint8_t*)'
     void setFont(const uint8_t *fontData);
          ^
*** [.pio\build\esp32\src\BrewManiac.cpp.o] Error 1
=============================================== [FAILED] Took 9.58 seconds ===============================================
Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:09.580
========================================= 1 failed, 0 succeeded in 00:00:09.580 ========================================= 
The terminal process "C:\Users\shingle\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

vitotai, 2022-03-07:
You must have wrong OLED library
Check the line 175 in OLEDDisplay.h

https://github.com/vitotai/BrewManiacEsp8266/blob/b959125d6e6eecb8307cd41a4131f3a8a483e2aa/lib/esp8266-oled-ssd1306-4.0.0/src/OLEDDisplay.h#L175

Your error message shows:
drawXbm(int16_t x, int16_t y, int16_t width, int16_t height, const uint8_t xbm);

It is:
void drawXbm(int16_t x, int16_t y, int16_t width, int16_t height, const uint8_t *xbm);

shingle01, 2022-03-07:
That's what I have:

[image]

vitotai, 2022-03-08:
You might try modify platform.ini
find  [env:esp32] section, and edit the "build_flags" to be like

build_flags =
    -DOLED_LCD=true

shingle01, 2022-03-12:
Tested and working.  Thanks.

closed 2022-03-12 by shingle01