vitotai/BrewManiacEsp8266 #105: Сука верни Русский язык!!!

jhnf, 2022-04-06:
Это по вашему называется "демократией", "толерантностью" и "свободой слова", я правильно понимаю? Удалить Русский язык по причине всеобщей русофобии, прям как фашисты с Евреями... Это называется ГЕНОЦИД!!! Хватит смотреть ваши однобокие СМИ, они там только разжигают русофобию и душат всех ложью!!! Посмотри уже даже некоторые ваши начинают прозревать, что не Россия виновата, а в первую очередь америка и европа, но их быстро в потоке лжи размывают!!! Не поддавайся русофобии, не будь ФАШИСТОМ!!! иначе травлю русских никак не назвать!

Philusha1983, 2022-04-06:
Рекомендую успокоиться, 
Вообще так демократия работает, так как российская федерация решила нарушать международные правила и вступила в войну против мирного населения, так и мы демократы всего мира используем наши демократические силы и исключаем рФ из международных структур. В том числе и тут...
Лично мне как русскоязычному человеку очень стыдно за то что происходит, вообще не понимаю российскую агрессию.
И понимаю что каждый граждан этого мира пытается сделать какие ты маленькие поступки которые могут как-то наказать рф

EroshKA, 2022-04-08:
[jhnf](https://github.com/jhnf), ты как раньше жил без русского языка?
В тот же день, когда Vito убрал поддержку русского языка ты сам что написал?
Vito владеет русским языком? Как он должен понять что работает или не работает после [таких писулек](https://github.com/vitotai/BrewManiacEsp8266/issues/14#issuecomment-1087230359)?
https://github.com/vitotai/BrewManiacEsp8266/issues/14#issuecomment-1087230359
Может совпадение, а может и прямое действие - не работает, значит удаляем.
Если хотите чтоб вас понимали, то тогда пишите внятно, раз надеетесь на google переводчик.
И после таких истерик, я, давший файлы локализаций, больше не хочу чтоб кто-то ими пользовался. Делайте сами.
[Philusha1983](https://github.com/Philusha1983), когда Израиль бомбит Сирию - вам не стыдно. Когда на Донбассе убивают русских людей вам стыдно за Россию, хоть и убивает армия Украины. Мы сами разберемся. Ваша однобокая демократия нам не нужна.

wzooroo, 2022-04-15:
https://github.com/EroshKA он убрал не потому что не работает, а тупо политика!

Philusha1983, 2022-10-11:
Рекомендую успокоиться,
Вообще так наша демократия работает, так как российская федерация решила
нарушать международные правила и вступила в войну против мирного населения,
так и мы демократы всего мира используем наши демократические силы и
исключаем рФ из международных структур. В том числе и тут...
Лично мне как русскоязычному человеку очень стыдно за то что происходит,
вообще не понимаю российскую агрессию.
И понимаю что каждый граждан этого мира пытается сделать какие ты маленькие
поступки которые могут как-то наказать рф

*Philip* :)