vitotai/BrewManiacEsp8266 #106: PID Autotune wrong Setpoint

Malord958, 2022-04-06:
In manual mode, starting autotune changes the target temperature to the current temperature. (setpoint=CurrentTemperature instead of setpoint=SettingTemperature). Tested with version 0.5.1 (bin file). 
maybe the cause of issue #93? behavior is the same
[image]
[image]
.

vitotai, 2022-04-08:
The target temperature of autotune is the temperature reading when you start, not the setting temperature.

Malord958, 2022-04-24:
OK sorry. The screenshots in the documentation made me think there was a issue.
[image]
https://github.com/vitotai/BrewManiacEsp8266/blob/master/docs/UserManual/3.ManualModeAndAutoTune.md

closed 2022-04-24 by Malord958