vitotai/BrewManiacEsp8266 #108: Multi Sensor Use cases

shingle01, 2022-05-02:
I'm sorry if this is the wrong place to post but I couldn't see an official forum attached to the project.

I'm a little confused about the use case for multiple temperature probes at "different stages".  My guess is it's targeted toward a multi-vessel setup?  I don't believe the current design helps me with the following 5 uses I have for temperature probes:

1. eBIAB vessel bottom
2. eBIAB vessel top
3. HLT for sparging
4. Counterflow chilled output
5. Recirculated cooling water reservoir

I'd like the mash PID to use both the eBIAB top/bottom (primary/aux) probes average.

I'm guessing Multisensor will support my HLT use case.

I'd like to monitor the chilled output, reservoir and vessel during cooling.

As a side note, I also have my own level sensor using a pressure sensor, a small air pump, and a dip tube to measure hydrostatic pressure.
[image]

Below is the Home Assistant sensor configuration which may help.  It includes temperature correction.  My pressure sensor measures in Pa.

```
  - platform: template
    name: "Vessel Volume"
    lambda: |-
      const float gravity = 9.8067;
      const float pi = 3.14159265359;
      const float maxpressure = 6000.0;
      const float radius = 0.198;
      const float deadspace = 0.12;
      const float lbsforce = 4.4482216152605;
      const float ZERO_CAL = 0.52401;
      const float pressure = (id(ads1115_a1).state - ZERO_CAL) / (4.0 / maxpressure);
      const float waterdensity = (999.83952 + (16.945176*id(vessel_temp).state) - ((7.9870401*powf(10,-3)) *  powf(id(vessel_temp).state, 2)) - ((46.170461*powf(10,-6)) *  powf(id(vessel_temp).state, 3)) + ((105.56302*powf(10,-9)) *  powf(id(vessel_temp).state, 4)) - ((280.54253*powf(10,-12)) * powf(id(vessel_temp).state, 5))) / (1 + ((16.897850*powf(10,-3))*id(vessel_temp).state));
      const float height = pressure / (waterdensity * gravity);
      const float volume = ((pi * powf(radius, 2) * height * 1000) + deadspace);
      if (volume > 1) return volume; else return 0.0;
    update_interval: 2s
    icon: 'mdi:cup'
    unit_of_measurement: 'L'
    accuracy_decimals: 2
```

vitotai, 2022-05-02:
Initially when I was using a Grainfather, stratification of mash temperature is concerned, and so multiple sensor support was developed. It happens to have the flexibility of supporting multiple vessels and RIMS. I've used it with a 2-vessel system and a system with RIMS.

I've thought of using average reading, but I am not sure how it will work. What maters most is the liquid part of mash where most enzyme is in.

shingle01, 2022-05-02:
Ah I see, as I suspected.

I always like to compare to what Kegland are doing as they appear to be the innovator when it comes to affordable products.

That being said BrewmanicEX in IMHO is a better, more flexible controller than their new Brewzilla gen 4 + rapt.  

Additional ds18b20's don't bother me but I don't see why BrewmaniacEX couldn't also be a Bluetooth gateway?

[image]

So you see their intention is for multiple sensors that happen to meet my use cases provided above.

closed 2022-06-06 by shingle01