vitotai/BrewManiacEsp8266 #111: Hopstand Temperature Conversion

frank-nobrega, 2022-06-18:
Last brew had a hopstand at 176F.  When the recipe loaded, it showed a Temp of 80.  It appears the Celcius temperature in the BeerXML did not get converted to Farenheit.  All other mash and boil temps were converted correctly.  I am using the ESP32 version.

vitotai, 2022-06-20:
According to BeerXML specification, 

"All temperatures will be measured in degrees Celsius."

http://www.beerxml.com/beerxml.htm

frank-nobrega, 2022-06-20:
Yes, Celsius is the BeerXML standard but in Brewmaniac you can choose to
use Farenheit.  It correctly converts mash and boil temps from the BeerXML
recipe import but the Hopstand temperature is not converted and it should
be.

frank-nobrega, 2022-06-24:
What is the cal file?

vitotai, 2022-06-24:
XML file. blame the auto correction.

vitotai, 2022-06-25:
It doesn't look like a valid BeerXML. You might try attaching the file you are using to import.

frank-nobrega, 2022-06-27:
Github would not allow upload of xml file.  I have changed the extension to .txt, you can change it to .xml to proceed
[Brewfather_BeerXML_ExtraSpecialBitterDavidHeath_20220610.txt](https://github.com/vitotai/BrewManiacEsp8266/files/8991852/Brewfather_BeerXML_ExtraSpecialBitterDavidHeath_20220610.txt)
.

frank-nobrega, 2022-07-01:
I posted the XML file as a .txt file a few days ago.

vitotai, 2022-07-03:
fixed.

FYI, I am not dedicated to this project only.

frank-nobrega, 2022-07-03:
Thank you so much for your effort on this project.  When the controller on
my very old Grainfather died it was just too much money to buy a
replacement so this project let me continue to use it with much more
functionality than the original controller.