vitotai/BrewManiacEsp8266 #113: pt100, 1000 sensors

ant1fr1z, 2022-07-20:
Is it possible to implement support for pt100, 1000 sensors?

vitotai, 2022-07-21:
Yes. But not for me. I don't have the hardware and requirement. 

I think, using ESP32 is more reasonable for more available PINs. A MAX31865 module should be helpful. 

https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/arduino-code

ant1fr1z, 2023-01-08:
> Yes. But not for me. I don't have the hardware and requirement.
> 
> I think, using ESP32 is more reasonable for more available PINs. A MAX31865 module should be helpful.
> 
> https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/arduino-code

Could you make support for these sensors? This is very important, they are smaller and sometimes they need to be used.

vitotai, 2023-01-09:
I've used MAX6675 in another project which needs to measure up to 250 degree Celsius.
It definitely can be done. However, I am busy these days and don't feel like fiddling the hardware setup which I won't use it.

vitotai, 2023-01-30:
You can try Max6675 with this branch:
https://github.com/vitotai/BrewManiacEsp8266/tree/PTsensor

pin assignment 
SCK 18
CS 23
MISO 19

That's the best I can do.