vitotai/BrewManiacEsp8266 #114: how install audio files in esp32

felipeserta, 2022-10-07:
The code runs perfectly in esp32 but i dont know how to install the audio files. I tried copy files to esp32 via http://bm.local:8008/filemanager in the same way I did with esp8266 without sucess and file names lost the first letter. How to solve this issue. Thanks

[image]

felipeserta, 2022-10-10:
Also, after saving a recipe file it doesnt apear in saved list for future use. Maybe the cause is the diference in file names without the first letter. I ll try to download to a different esp32 board.

thnaks

vitotai, 2022-10-11:
Which version do you use? 
I know there are issues about file names because of difference between SPIFFS and LITTLEFS. I've spent some time on this.

felipeserta, 2022-10-11:
I´m using last one, version 0.5.1. Yesterday I tried new board and update firmware esp32 to 5.2.0 without success. I tried to upload files creating a data folder and upload via platformio but in file manager lost the first letter again. 
thanks for any help.

vitotai, 2022-10-14:
I am sorry by neglecting BrewManiacEx in this issue when I had fixed this in BrewPiLess.
Please check the latest update.

felipeserta, 2022-10-14:
After a tried the new code the first letter apears again but without R/ at recipes files.

Before that I did the changes below and worked at filemanager
[image]

At ESPAUpdateServer.cpp I made this changes 
static void handleFileList(void) {
....
 #if UseLittleFS
    output += entry.name();
    #else
//new code
     #if ESP32
    	output += String(entry.path()).substring(1);
      #else
    output += String(entry.name()).substring(1);
		 #endif									
    #endif
//end of new code
    output += "\"}";

But I still can´t retry recipes files
[image]
Serial print shows at this time
[image]

I didn´t try brewlogs yet but show this
[image]

Thanks a lot

felipeserta, 2022-10-14:
Brew logs are being saved, but still get this empty logs, I did not remember in other versions.
[image]
thanks

vitotai, 2022-10-15:
I tried and found out that the latest SPIFFS of ESP32 framework might have issue with directory listing.
I inserted a test code to create a file with subdirectory, but it appeared in root directory.

It might take a while to solve this issue. Well, It might never be solved.

I tried with LittleFS, and it seems to work.

BTW,
I have issues with SPIFFS on BrewPiLess after running for 3 months. I am now switching to LittleFS and running.

felipeserta, 2022-10-15:
Sure looks like new issues with spiffs libs. With the changes above it behaves like previous version and everything is working filemanager, recipes and logs with esp32. Now I´ll take a look at audio files.
Little FS looks like a better and cleaner code.
Thanks

felipeserta, 2022-10-15:
After uploading sounds.json and sounds.mp4 I forgot to configure webpage to allow sound. All fine by now.

thks