vitotai/BrewManiacEsp8266 #116: LCD web page like brewpiless

raul811, 2022-10-29:
Great project I use both bm and bpl .
Can you include in next version a lcd web page like in brewpiless with bottuns direct from web ui because I use the project with my phone and active Google  assistant because I am blind guy .
The web ui lcd in brewpiless works great with Google assistant , reads everything it is important for me and it's very easy to use.
 Thanks!

vitotai, 2022-10-30:
I don't really understand you.
Is the lcd web page you mentioned the dedicated "/lcd" page of BrewPiLess?

There is in fact a mirrored display of LCD in BM web UI, even with buttons. It will be very easy to extract LCD part to make a dedicated page. However, derived from Open ArdBir, BrewManiacEx uses custom characters for pump and heating status. I am not sure how they will be readout in Google assistant.

I am happy to help if possible, but some more detailed requirements are needed.

raul811, 2022-10-30:
The dedicated lcd from bpl is read by the voice assistant from my phone but in bm web ui page reads  only the buttons name.
If it is possible a dedicated lcd or web ui with a minimal information: temp.set,temp.primary,temp.aux
,time and the four buttons are very use full.

vitotai, 2022-10-30:
The LCD display is generated by drawing instead of pure text because there are custom characters, like temperature unit symbol.
I can create a dedicated page that display something like this.
1st part:LCD mirror excluding special symbols.
2nd part:Four buttons.
3rd part:the information about temperature, set, and time.
4th part:hint, optional.

For example:

AUTO --> Mash 1/1
59.65°C 65.00Set
01:00:00
UP* *DWN PmPus STP
  (four buttons with labels)
HEAT ON
PUMP ON
Temperature 59.66 °C
Set 65.00 °C
PWM -- %
Time 01:00:00
Automation Mash Step(1) 65°C 60min
Next: Mash Out 76°C 10min

How do you think?

raul811, 2022-10-30:
Great and thank u again.

vitotai, 2022-10-31:
Please try uploading the attached file as "console.htm.gz" and access it as "console.htm", if possible.

Any suggestions, recommendations, and opinions are welcome. 

BTW, have you try audio prompt?
https://github.com/vitotai/BrewManiacEsp8266/tree/master/audio

[console.htm.gz](https://github.com/vitotai/BrewManiacEsp8266/files/9901295/console.htm.gz)

raul811, 2022-10-31:
I uploaded the file and I opened it the console.hcm it is working but aren't assigned values on temps , heat,pump,time.
  I can't see values.

raul811, 2022-10-31:
[image]
U

raul811, 2022-10-31:
[image]

vitotai, 2022-11-01:
remove :8008 from the address.
It should be
http://192.168.1.52/console.htm

raul811, 2022-11-01:
It is working very well .In the future I want try the audio prompt .
And for a suggestion  if u can program a hardware button: when the button is pressed the bm answer with the buzzer  and says the set point , actual temp,
Actual time and the brewing steps.
 Thanks for everything it's great

vitotai, 2022-11-02:
To implement audio, ESP32, or an extra component, would be needed for its DAC, Digital to Analog Conversion, capability.
Moreover, TTS, Text To Speech, is the key to implement this feature. However, I am not sure if ESP32 have enough RAM and/or flash memory to handle it.

closed 2022-11-03 by raul811