vitotai/BrewManiacEsp8266 #117: Can you use two heaters and sparge

jimhiggs, 2023-02-11:
I have a 2 element single vessel biab type system (graninfather s40) that I am converting to BrewManiac. I also have a Coffee urn hot water heater that I am using to heat for sparge. I configured for both spargeheatersupport and secondaryheatersupport. On the web interface it looks like its working but when I look at pins.h line 59 for SpargeHeaterSupport it is writing to AuxHeatControlPin the same as line 52 for SecondaryHeaterSupport. I see no SecondaryHeatControlPin defined. Is that correct?

Can I define a different pin so both can be used or am I just misunderstanding this?

Also I dont need the exclusivity of Main heat and sparge heat as my sparge is running on a separate 120v circuit from my main heaters which are on 240v. I was looking at the code in BrewManiac.cpp for where it is disabling Sparge heater when other heaters are active. Would it be reasonably simple to change this functionality or should I just leave it alone?

vitotai, 2023-03-02:
No. Limited number of available PINs limited my minds. I haven't thought of using both at the same time.
Given enough pins, it should be possible. However, there might be some surprise....

jimhiggs, 2023-03-02:
Thanks for the reply. I have two elements in my kettle and a separate sparge heater. I can combine the two kettle elements but I had planned to have  a relay on one and the SSR on the other. I am using a 30pin eps32 so guess there would be enough pins on that should I want to pursue this. Brewmaniac is a great alternative to the useless controller on the Grainfather S40 so thanks for all your work.

vitotai, 2023-03-03:
I think it should work, but some test is definitely needed.
Small change of the code is ongoing, and I will do some simple test.
You might need to dry run for verification.

shingle01, 2023-03-17:
Having both the following in the build flags:

```
-DSecondaryHeaterSupportEnabled=true 
-DSpargeHeaterSupportEnabled=true
```
The compiler is not loving it:

```
#if SecondaryHeaterSupport == true
,
_PrimaryHeaterSymbol, //10
_RevPrimaryHeaterSymbol, //11
_SecondaryHeaterSymbol, //12
_RevSecondaryHeaterSymbol //13
#endif
```
[image]

shingle01, 2023-03-17:
Nevermind, I accidentally tried to compile the build options on esp32-oled.  

As I have a standard 2004 20x4, I found the code successfully compiles the code.  I'll test the dual heater / sparge capabilities soon.