vitotai/BrewManiacEsp8266 #118: No way to skip step or end brew manually and Heat@Pump for mash only

jimhiggs, 2023-02-25:
I found BrewManiac when looking for non pi alternatives for Craftbeerpi or Brewpi. I bought a Grainfather S40 and want better heat control. Looked at others but BrewManiac seems to be what I was looking for.
A couple of things that I miss. 
1. There seems to be no way to skip a step when in Auto.
2. The only way I can see to stop a brew session or restart is to pull the plug is that right?
3. For both it would be nice if these buttons existed when in Pause.
4. Heat@Pump is a really useful feature. In my 3 vessel system running CraftbeerPi  I use a probe in the recirc tube to control mash temp and when turning off the pump have managed to overheat the mash.  But I don't run the pump during boil so I would like to implement Heat@Pump only during mash modes.
I will start looking at the code to see if I can adapt it myself.

frank-nobrega, 2023-03-02:
In the settings section of BrewManiac you can scroll down to Pump on Boil and turn it off.

jimhiggs, 2023-03-02:
@frank-nobrega Yes I know but once running in manual or auto you cant edit settings can you
@vitotai Thanks for the answer to 1 & 2. Dont know how I missed this. 
  3. While in pause only one button is defined (exit) I was thinking 1 could be used for skip step and 1 for Quit to main menu. But your answer to 1 & 2 provides the function.
  4.  I will give this a try.

jimhiggs, 2023-03-03:
@vitotai  in 4 above I think you meant `if(_isPIDMode && readSetting(PS_HeatOnPump)){ ...`
 (not !_isPIDMode) 
Works great.

vitotai, 2023-03-03:
@jimhiggs 
My bad. You are 100% right. I might have had too much beer last night.