vitotai/BrewManiacEsp8266 #120: How to make changes to the web interface

Symbol95, 2023-03-28:
Hello!
I need to make changes to the web interface. I want to change names, I want to change the names, for this I make changes in html /src/locales/english.json, save the file, compile the project, load the bin file into the controller ESP8266.
I open the web interface and see no changes, refresh the page, clear browser cache, I clear the browser cache but nothing changes.
What am I doing wrong?
Perhaps you need to make changes to some other file?

vitotai, 2023-04-24:
You will need some basic programming knowledge.
https://github.com/vitotai/BrewPiLess/tree/master/htmljs

Symbol95, 2023-05-03:
Installed mpn and gem, run grunt build in html directory.
A dist folder appeared in the project, then I compile and send it to the controller, I go to the site and see no changes.
What else needs to be done?

vitotai, 2023-05-03:
refresh the browser.

Symbol95, 2023-05-03:
Of course, I restarted the browser and cleared the cache

vitotai, 2023-05-04:
How did you send it to the controller? What exactly did you send?

Symbol95, 2023-05-04:
In the platformio below I click on the arrow and send it to the controller.
Need to do it differently?

vitotai, 2023-05-04:
Well. You have to convert the bm.htm.gz file into a C header file, and put it in "locales/" in the right name. There is no automation for this.
Or, you can upload the result bm.htm.gz by file manager, which is http://[address of the controller]:8008/filemanager

Symbol95, 2023-05-04:
Thanks for the answer.
I will try