vitotai/BrewManiacEsp8266 #124: Compile error with version 0.5.2

wimdm1, 2023-10-14:
When I try to compile the version 0.5.2 I get the following error:

**src\EmbeddedFiles.cpp:34:1: fatal error: english_bm_htm.h: No such file or directory_**

Version 0.5.0 goes fine but I can not see any difference in 'EmbeddedFiles.cpp' of both versions except that the russian_bm_htm.h and resources_ru.h files are missing in version 0.5.2. under the directory locales

Haggenti, 2024-01-13:
Hi, i had the same issue. Solved in checking options in platformIO project directory options, 
Include_dir must be locales/
Good day.

andrej1290, 2024-02-08:
I have same problem from version 0.5.3. 
Can you give me more information about what this means:
"Solved in checking options in platformIO project directory options, Include_dir must be locales/ "

Thanks

vitotai, 2024-02-11:
Isn't that included in platformio.ini?