vitotai/BrewManiacEsp8266 #126: Compile error. Version 0.5.3

Omnn82, 2024-03-11:
Can you tell me what causes the following error when compiling the firmware? Version 0.5.3.

collect2.exe: error: ld returned 1 exit status
*** [.piobuilddistillingfirmware.elf] Error 1

```
*  Executing task in folder BrewManiacEsp8266: C:\Users\user\.platformio\penv\Scripts\platformio.exe run 

Processing distilling (platform: espressif8266@~4.2.1; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.2.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.3
|-- ESP Async WebServer @ 1.2.3+sha.43381ab
|-- EEPROM @ 1.0
|-- OneWire @ 2.3.2
|-- PID_AutoTune_v0
|-- PID_v2
|-- ArduinoOTA @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ESP8266WiFi @ 1.0
|-- ESPAsyncTCP @ 1.1.3
|-- Hash @ 1.0
|-- LittleFS @ 0.1.0
|-- EspSoftwareSerial @ 8.0.1
|-- Wire @ 1.0
|-- ESP8266HTTPUpdateServer @ 1.0
|-- ESP8266WebServer @ 1.0
|-- DNSServer @ 1.1.1
|-- PCF8574 - library for ESP8266 @ 1.0.0
|-- ESP8266_SSD1306 @ 4.0.0
|-- LiquidCrystal_I2C_V3-master
|-- ESP8266HTTPClient @ 1.2
|-- ESP8266httpUpdate @ 1.3
Building in release mode
Warning! '-Wl,-T' option for specifying linker scripts is deprecated. Please use 'board_build.ldscript' option in your 'platformio.ini' file.
Linking .pio\build\distilling\firmware.elf
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z11pidGetValuei+0x8): undefined reference to `_Z15readSettingWordi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z11pidGetValuei+0xc): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o: in function `_Z11pidGetValuei':
BrewManiac.cpp:(.text._Z11pidGetValuei+0x52): undefined reference to `_Z15readSettingWordi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: BrewManiac.cpp:(.text._Z11pidGetValuei+0x96): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z21wiUpdatePrimarySensorhh+0x0): undefined reference to `_Z13updateSettingih'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o: in function `_Z14btnReadButtonsv':
BrewManiac.cpp:(.text._Z14btnReadButtonsv+0xd9): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z25calTemperatureCalibrationv+0x20): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z25calTemperatureCalibrationv+0x43): undefined reference to `_Z15readSettingWordi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z25calTemperatureCalibrationv+0x5c): undefined reference to `_Z15readSettingWordi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o: in function `_Z25calTemperatureCalibrationv':
BrewManiac.cpp:(.text._Z25calTemperatureCalibrationv+0x68): undefined reference to `_Z15readSettingWordi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: BrewManiac.cpp:(.text._Z25calTemperatureCalibrationv+0x72): undefined reference to `_Z15readSettingWordi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: BrewManiac.cpp:(.text._Z25calTemperatureCalibrationv+0x128): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z19wiUpdateSettingWordis+0x0): undefined reference to `_Z17updateSettingWordit'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o: in function `_Z19wiUpdateSettingWordis':
BrewManiac.cpp:(.text._Z19wiUpdateSettingWordis+0x16): undefined reference to `_Z17updateSettingWordit'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o: in function `_Z27wiUpdateCalibrationOfSensorhh':
BrewManiac.cpp:(.text._Z27wiUpdateCalibrationOfSensorhh+0x25): undefined reference to `_Z13updateSettingih'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o: in function `_Z17loadSensorSettingv':
[... 80 lines cut ...]
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z20autoModeEnterBoilingv+0x21): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z20autoModeEnterBoilingv+0x65): more undefined references to `_Z11readSettingi' follow
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z18autoModeAutoResumev+0x28): undefined reference to `_Z15readSettingWordi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o: in function `_Z28autoModeMashingStageFinishedv':
BrewManiac.cpp:(.text._Z28autoModeMashingStageFinishedv+0x2a): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: BrewManiac.cpp:(.text._Z28autoModeMashingStageFinishedv+0x54): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z32autoModeMashInAskContinueHandlerh+0x19): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiac.cpp.o:(.text._Z32autoModeMashInAskContinueHandlerh+0x26): undefined reference to `_Z11readSettingi'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\distilling\src\BrewManiacWeb.cpp.o:(.text._ZN13BrewManiacWeb11getSettingsER6String+0xa0): undefined reference to `_Z11readSettingi'
[... cut ...]
================================================================================== [FAILED] Took 3.92 seconds ==================================================================================

Environment    Status    Duration
-------------  --------  ------------
distilling     FAILED    00:00:03.924
============================================================================ 1 failed, 0 succeeded in 00:00:03.924 ============================================================================ 

 *  The terminal process "C:\Users\user\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

dmcaetano1, 2024-03-22:
I'm facing exactly same problem, with the same result...

vitotai, 2024-03-24:
I couldn't reproduce. 
It seems like a compiler relate issue. There is no '_Z13commitSettingv' function but 'commitSetting'. The compiler seems to generates another symbol for some reasons.

Omnn82, 2024-03-27:
On version 0.5.2 there is no such problem...

vitotai, 2024-03-30:
my bad. One file was missed.

dmcaetano1, 2024-03-30:
Now, everything is fine here. Thanks Vitotai!